rwaldron/johnny-five #1357: MPU6050 and RaspberryPI

artus9033, 2017-06-26:
I'm using node v8.0.0 with exactly the same code as in the mpu6050 example and with the same pinout (using the Invensense mpu6050), but the change and data events never fire. If I use setInterval to print the values manually, they'll always be 0. The output in the console looks like this:

`1498461023666 Available /dev/ttyUSB0  
1498461023731 Connected /dev/ttyUSB0  
1498461027536 Repl Initialized  
>> 0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0
0 0 0

(To exit, press ^C again or type .exit)
>> 0 0 0

1498461047866 Board Closing.  
`

I tried the newest (v0.11) and v0.9.58, but such a change didn't bring me any effects. Do you know what's wrong?

fivdi, 2017-07-02:
@artus9033 Is the MPU6050 connected to and Arduino UNO and the Arduino UNO connected to a Raspberry Pi via USB? And the Johnny-Five program is being run on the Raspberry Pi?

artus9033, 2017-07-02:
Yes, exactly. I've just discovered that it may've been something related to my mpu, because when i connected the mpu directly to the rpi and ran i2cdetect -y 1, and there magic happened - sometimes it was seen (I got output on the 0x68 address) and sometimes the rpi didn't see any i2c devices - I had to refresh several times with approx. 3/4 chance of the module not being detected. What helped was to plug the AD0 to 3v to change the address to 0x69 and then it worked - johnny five started working instantly & also the rpi i2cdetect has started to see the module continously. (I think I'll close the thread, cause it was not johnny five to blame. Thank you for help)

closed 2017-07-02 by artus9033