rwaldron/johnny-five #1358: Analog pin mapping labels appear to be sequential instead of looked up

ricklon, 2017-07-04:
Analog pin mapping labels appear to be sequential instead of looked up.

These are the analog pins for the Fubarino Mini board.
```
     analogPins: [ 0, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13 ],

```
I can declare a sensor with:

```
  const potA = new five.Sensor({ pin: 'A0' });
  const potB = new five.Sensor({ pin: 'A1' });

```
The results is A0 is pin 0 and A1 is pin 1.  The result should be A1  is pin 3.

rwaldron, 2017-07-05:
Thanks for filing an issue. The assumption described here is only partly correct. The digit part of `"A1"` is only sequential in that it maps to an _index_ in `analogPins` (which is a property of the IO plugin, and I assume we're talking about "default" Johnny-Five and Firmata.js). Firmata.js tracks analog pins by an index that maps to the actual integer value name of an analog pin: 

- https://github.com/firmata/firmata.js/blob/786c23cfeb6b600adcb25f5571d8c9e9c85946d5/lib/firmata.js#L799
- Mark local pin record as "reporting analog" https://github.com/firmata/firmata.js/blob/786c23cfeb6b600adcb25f5571d8c9e9c85946d5/lib/firmata.js#L1625
- Send message to remote board instructing it to start reporting analog values from this `pin` (which is actually an _index_). https://github.com/firmata/firmata.js/blob/786c23cfeb6b600adcb25f5571d8c9e9c85946d5/lib/firmata.js#L1626
- On the board, StandardFirmata (and any of its variants) will convert the pin _index_ to the right integer value name, for the right board: 
  - https://github.com/firmata/arduino/blob/f13a61bce62406fc182083cf0c527f25b3ef38c1/Boards.h#L161
  - https://github.com/firmata/arduino/blob/f13a61bce62406fc182083cf0c527f25b3ef38c1/Boards.h#L177
  - https://github.com/firmata/arduino/blob/f13a61bce62406fc182083cf0c527f25b3ef38c1/Boards.h#L193
  - https://github.com/firmata/arduino/blob/f13a61bce62406fc182083cf0c527f25b3ef38c1/Boards.h#L218
  - https://github.com/firmata/arduino/blob/f13a61bce62406fc182083cf0c527f25b3ef38c1/Boards.h#L403
 - And even for the fubarino mini: https://github.com/firmata/arduino/blob/f13a61bce62406fc182083cf0c527f25b3ef38c1/Boards.h#L594 (which appears to have no special rules in place for converting an _index_ into a valid analog pin number... which is problematic for StandardFirmata*'s analog read phase: https://github.com/firmata/arduino/blob/f13a61bce62406fc182083cf0c527f25b3ef38c1/examples/StandardFirmataPlus/StandardFirmataPlus.ino#L832-L838 (which maps to https://github.com/firmata/arduino/blob/f18eb7519246a78563c3cae638cb11d01568fcbb/FirmataMarshaller.cpp#L245-L255). Apparently this issue was well known to the contributor of that Boards.h definition: https://github.com/firmata/arduino/blob/f13a61bce62406fc182083cf0c527f25b3ef38c1/Boards.h#L583. I think the correct fix is to change the [FubarinoMini definition of `PIN_TO_ANALOG`](https://github.com/firmata/arduino/blob/f13a61bce62406fc182083cf0c527f25b3ef38c1/Boards.h#L580-L596) to: 

```c
#define PIN_TO_ANALOG(p)        ((p) == 0 ? 0 : ((((p) >= 1 && (p) <= 11)) ? ((p) + 2) : (p)))
```

@soundanalogous should check my work here.

ricklon, 2017-07-05:
Thank you for the detailed feedback. I'll try that rule out and confirm.
Then I'll get that update added.

-_Rick

ricklon, 2017-07-05:
I've dug into it further and have confirmed that settings in Firmata are incorrect. I wrote a test sketch that tests the functions old and proposed fixed.
```
#define PROP_IS_PIN_ANALOG(p)        ((p) == 0 || ((p) >= 3 && (p) <= 13) || (p) == 20)
#define PROP_PIN_TO_ANALOG(p)        ((p) == 0 ? 0 : ((((p) >= 1 && (p) <= 11)) ? ((p) + 2) : (p)) )
#include <Servo.h>
#include <Wire.h>
#include <Firmata.h>

void setup() {
  pinMode(PIN_BTN1, INPUT);
  Serial.begin(9600);
}

void loop() {
  static uint8_t p1 = digitalRead(PIN_BTN1);
  if (digitalRead(PIN_BTN1) != p1) {
    p1 = digitalRead(PIN_BTN1);
    if (p1 == HIGH) {
      Serial.println("Firmata: Current");
      for (int ii = 0; ii < 22; ii += 1) {
        Serial.printf("A%d, toPin: %d, isPinAnalog: %d\n", ii, PIN_TO_ANALOG(ii), IS_PIN_ANALOG(PIN_TO_ANALOG(ii)));
      }
      Serial.println("Firmata: Proposed");
      for (int ii = 0; ii < 22; ii += 1) {
        Serial.printf("A%d, toPin: %d, isPinAnalog: %d\n", ii, PROP_PIN_TO_ANALOG(ii),
                      PROP_IS_PIN_ANALOG(PROP_PIN_TO_ANALOG(ii)));
      }
    }
  }
}
```
I'll contact them to add these corrections. Also, I'll test issue with the proposed fixes.

ricklon, 2017-07-05:
I've got a test version of StandardFirmata with an update Boards.h. This is the code I'm using to test Johnny-Five with. It's really brute force. Maybe, a better test.:
```
const config = require('config');
const logger = require('winston');
const five = require('johnny-five');

const PIN_LED1 = 1; // Fubarino_mini pin 1

// const A0 = 0;
const board = new five.Board({
  port: config.get('port'),
});

board.on('ready', function() {
  //const led = new five.Led(PIN_LED1);
  // Strobe the pin on/off, defaults to 100ms phases
  //led.strobe();
  let a0 = new five.Pin({
    pin: "A0"
  });
  let a1 = new five.Pin({
    pin: "A1"
  });
  let a2 = new five.Pin({
    pin: "A2"
  });
  let a3 = new five.Pin({
    pin: "A3"
  });
  let a4 = new five.Pin({
    pin: "A4"
  });
  let a5 = new five.Pin({
    pin: "A5"
  });
  let a6 = new five.Pin({
    pin: "A6"
  });
  let a7 = new five.Pin({
    pin: "A7"
  });
  let a8 = new five.Pin({
    pin: "A8"
  });
  let a9 = new five.Pin({
    pin: "A9"
  });
  let a10 = new five.Pin({
    pin: "A10"
  });
  let a11 = new five.Pin({
    pin: "A11"
  });

  logger.log('info', `A0: ${a0.pin}`);
  logger.log('info', `A1: ${a1.pin}`);
  logger.log('info', `A2: ${a2.pin}`);
  logger.log('info', `A3: ${a3.pin}`);
  logger.log('info', `A4: ${a4.pin}`);
  logger.log('info', `A5: ${a5.pin}`);
  logger.log('info', `A6: ${a6.pin}`);
  logger.log('info', `A7: ${a7.pin}`);
  logger.log('info', `A8: ${a8.pin}`);
  logger.log('info', `A9: ${a9.pin}`);
  logger.log('info', `A10: ${a10.pin}`);
  logger.log('info', `A11: ${a11.pin}`);
  //  logger.log('info', `A12: ${a12.pin}`);
  //  logger.log('info', `A13: ${a13.pin}`);
  //  logger.log('info', `A14: ${a14.pin}`);

  board.repl.inject({
    a0: a0
  });
});
```

Results:
```
info: A0: 0
info: A1: 1
info: A2: 2
info: A3: 3
info: A4: 4
info: A5: 5
info: A6: 6
info: A7: 7
info: A8: 8
info: A9: 9
info: A10: 10
info: A11: 11
```

Still has wrong pin mapping.

rwaldron, 2017-07-07:
The value of `.pin` will be whatever Johnny-Five translates it to, which is going to be an index, so those values are correct.

ricklon, 2017-07-07:
Can I get a hold of the parent object of .pin in the repl and examine it in detail?

rwaldron, 2017-07-07:
`.pin` is just an number (integer) that corresponds to whatever Firmata.js wants it to be. The changes that i recommend for to `PIN_TO_ANALOG()` will not be observable from the JS side

rwaldron, 2017-07-07:
I've just ordered one of these boards so that I can work on this directly, I'll let you know once it's arrived.

rwaldron, 2017-07-12:
I've received my Fubarino Mini and started working to debug this issue. First thing to report: my earlier suggestion was not correct. I'm seeing unexpected replies at the firmata.js level (ie. no matter what pin I try to do an analogRead from, it always comes back as pin 0) and need more time to test and investigate.

soundanalogous, 2017-07-12:
Try adding an `#error "get's here"` statement to the Fubarino block in Boards.h to make sure it's actually picking up the constants for that board and not another block that may be defined at a general architecture level.

rwaldron, 2017-07-12:
Thanks, will do.

ricklon, 2017-07-17:
If there's any questions about the Fubarino let me know. Happy to help. This is awesome that you are checking out this problem to such detail.

closed 2018-08-28 by stale[bot]