rwaldron/johnny-five #1360: Altimeter class: why am I not seeing a change?

drewfurgiuele, 2017-07-05:
Hey everyone, sorry if this isn't the place for this: I have both a BMP280 and BMP180 wired up and I can use the example code provided, but I am never seeing my values change. The pressure/barometer values on the BMP280 change but I can't quite get any values to change on the actual altitude. I've raised and lowered the sensors by 20 - 50 feet in the air so far.

I also tested the sensors on stick arduino sensors and I was able to see small changes. Does anyone know why these values aren't changing?

rwaldron, 2017-07-07:
> I can use the example code provided,

Which code is that?

dtex, 2018-01-01:
Hi @drewfurgiuele it's been a while, but nobody wants to close this if you still need help. Is your issue solved? If not can you share which code you used?

dtex, 2018-01-16:
Hi @drewfurgiuele 

I'm going to close this since we haven't heard back. Feel free to re-open if necessary.

closed 2018-01-16 by dtex