rwaldron/johnny-five #1361: Cannot add a new segment to an existing animation

jbdesbas, 2017-07-09:
Hello,

I don't understand how to enqueue segments of animation. When I put a new segment with animation.enqueue(), then the entire animation is replaced by the new segment.

Can you help me please ?

Here is my code, only the leg of the second segment is moving : 
```
m.tap=function(tapleg){
	anim.enqueue({
		target:s.r1f,
		duration:300,
		fps:25,
		cuePoints:[0,0.5,1],
		keyFrames:[null,{degrees: 50,easing:easeInOut},{degrees: 90,easing:easeInOut}]
	});
	anim.enqueue({
		target:s.r2f,
		duration:300,
		fps:25,
		cuePoints:[0,0.5,1],
		keyFrames:[null,{degrees: 50,easing:easeInOut},{degrees: 90,easing:easeInOut}]
	});
}
```

>Board : Arduino ProMini 5V 16Mhz (ATmega 328)
>Johnny5 : johnny-five@0.11.3
>NodeJS : v4.2.6

dtex, 2017-07-09:
Is it your intent that these play at the same time?

jbdesbas, 2017-07-09:
Hello,
No, i want the second one play after. Idea ?

dtex, 2017-07-09:
I do think you've found a bug. While I track that down, I have a tip for making this work and be a little less code too...

````
s.femurs = new five.Servos([s.r1f, s.r2f]); // I'm guessing these are femurs

m.tap=function(tapleg){
	anim.enqueue({
		target:s.femurs,
		duration:600,
		fps:25,
		cuePoints:[0, 0.25, 0.5, 0.75, 1],
		keyFrames:[
                     [null,{degrees: 50,easing:easeInOut},{degrees: 90,easing:easeInOut}, null, null],
                     [null, null, null,{degrees: 50,easing:easeInOut},{degrees: 90,easing:easeInOut}]
                ]
	});
}
````
You can group any collection of Servos any way you like so creating a femurs group for an animation like this where you're just moving all the femurs makes sense. Servos can be in any number of groups.

jbdesbas, 2017-07-09:
Ok thank you, it is working. But I really don't understand why
animation.enqueue() don't add a segment after the existing one :-/

dtex, 2017-07-09:
> I really don't understand why animation.enqueue() don't add a segment after the existing one

It's a bug.

jbdesbas, 2017-07-09:
OK, for information, I use "PCA9685" controller for servo

dtex, 2017-07-09:
Cool. Just FYI, another user has had some problems with sending multiple, concurrent commands to their PCA9685. It could turn out to be an issue for you too #1340

dtex, 2017-07-09:
We need to check and see if their is a segment already running before call next() [here](https://github.com/rwaldron/johnny-five/blob/master/lib/animation.js#L158-L160)

PR #1376 (2017-08-18): Fix support for synchronous animation queues
Fixes #1361
commit af8494b8 (2017-08-18): Fix support for synchronous animtaion queues

Fixes #1361
closed 2017-08-18 by rwaldron