rwaldron/johnny-five #1362: motozero - pwm errors

nooitaf, 2017-07-09:
I'm struggeling getting the motozero to work with johnny-five. It complains about ports not being able to do PWM, but using GPIO instead of PWM is "the feature" of the motozero. I wonder if it's even possible to use all 4 motors with johnny-five.

I did some searching but i guess i missed that one page that explains it all. All i could find was python and or arduino related which both is kind of useless tbo. 

I used this diagram (https://pinout.xyz/pinout/motozero#) and used the Wiring-Pi numbers for motor1.

```js
motor = new five.Motor({
  pins:[5,2],
  pin: 21
})
```
```text
Error: Pin "21" does not support mode "3"
```

dtex, 2017-07-09:
> It complains about ports not being able to do PWM

We've always assumed that at least one of the pins is PWM for controlling speed so Johnny-Five would need to be modified to allow for digital pins to be used. This is the first request I've seen for that.

I don't have any real experience with Pi's. Perhaps @nebrius could offer some advice on how to get around this issue?

> using GPIO instead of PWM is "the feature" of the motozero

Sounds like the marketing department got involved in the product description. Any motor controller can be wired to digital pins. You just lose speed control. That's not something unique to this device.

> I wonder if it's even possible to use all 4 motors with johnny-five

With the right hardware you can control hundreds of motors with Johnny-Five.

So the question I have is, should we add support for non-pwm (On/Off) motors? If so, how should we modify the API. Something like this perhaps:

````
motor = new five.Motor({
  pins:[5,2],
  isPWM: false // defaults to true
});
````

On a tangent, there is an error in your configuration. You should only have ```pins``` or ```pin```, not both. But that's not going to solve your problem.

nooitaf, 2017-07-10:
> Sounds like the marketing department got involved in the product description. 

I guess the marketing department meant that it's 4 motors with speed control. I might have emphesised it a bit in my mind :D 

I'm also not sure if 2 pins would be enough for the motozero, there is still the `enable` pin.

This is the python example:
```python
from gpiozero import Motor, OutputDevice
from time import sleep

motor = Motor( 24 , 27 )
motor_enable = OutputDevice( 5 , initial_value=1 )

motor.value = 0.5 # half speed forwards
motor.value = -0.5 # half speed backwards
motor.value = 0 # stop
```

Maybe something like this:
```js
motor = new five.Motor({
  pins: {
    dir: 24,
    cdir: 27,
    enable: 5
  }
});
```

nebrius, 2017-07-10:
If you set the [enableSoftPwm](https://github.com/nebrius/raspi-io#new-raspioptions) flag in the Raspi IO constructor to `true`, then all pins will work with PWM. It won't be _quite_ as accurate as hardware PWM, and may increase CPU usage, but it will work.

dtex, 2017-07-12:
@nooitaf I'm kinda guessing here, but I think it should be
````
motor = new five.Motor({
  pins: {
    pwm: 24,
    dir: 27,
    enable: 5
  },
  invertPWM: true
});
````

nooitaf, 2017-07-13:
Awesome @nebrius, that sound exacly like the option needed. But i still get the pwm error :\ 
I would assume it shouldn't show that exact error anymore, so there might be something else i'm missing..

```js
const Raspi = require('raspi-io');
var five = require('johnny-five');
// const pwm = require('raspi-pwm');
var board = new five.Board({
  io: new Raspi({
    enableSoftPwm: true
  })
});

board.on("ready", function() {
  var motor = new five.Motor({
    pins: {
      pwm: 'P1-18',
      dir: 'P1-13',
      enable: 'P1-29'
    },
    invertPWM: true
  });

  motor.forward(100)
});
```
```txt
Error: Pin "P1-18" does not support mode "3" 
at Raspi.value (/home/pi/motortest/node_modules/raspi-io/lib/index.js:439:15)
...
```

package.js deps
```js
"dependencies": {
    "fs": "0.0.1-security",
    "http": "0.0.0",
    "johnny-five": "^0.10.13",
    "pi-gpio": "0.0.8",
    "raspi-io": "^6.2.0",
    "serialport": "^4.0.7",
    "socket.io": "^1.7.4"
  }
```

nebrius, 2017-07-13:
Ah, I see what it is now. In the dependencies section of your package.json, you have `"raspi-io": "^6.2.0"`, which will resolve to version 6.2.0. However, the `enableSoftPwm` flag was not [introduced until version 7.1.0](https://github.com/nebrius/raspi-io/blob/master/CHANGELOG.md#710-2016-11-29), so if you update to the latest version (now at v8.0.1), then it should work.

nooitaf, 2017-07-13:
Nice, that was indeed the issue, my bad. I blindly trusted `npm update`, guess thats not how it works. The error is gone now. But i still did't get it to work. Nothing happened :\ I figured hardware maybe?

I realised by testing with the python script again and multimetering the terminals that motor1 and motor2 are not getting any voltage at all. I guess one of my controllers is broken. So i moved the 2 motors i want to use to terminal 3 and 4 and they worked  with the script. Hardware works, check.

After going back to node and setting the pins to motor3 and using the  [motor-enable-example](http://johnny-five.io/examples/motor-enable/) @dtex suggested, but i still got silence...

Of course i forgot:
```js
motor.enable();
```
Wroooom!
It finally works ! \o/ Much thanks!!!

Here my example for all motors for later stumblers
```txt
"johnny-five": "^0.10.13",
"raspi-io": "^8.0.1",
```
```js
var Raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new Raspi({
    enableSoftPwm: true
  })
});

board.on("ready", function() {
  var motors = new five.Motors(
    [
      // motor 1
      {
        pins: {
          pwm: 5,
          dir: 2,
          enable: 21
        },
        invertPWM: true
      },
      // motor 2
      {
        pins: {
          pwm: 22,
          dir: 3,
          enable: 0
        },
        invertPWM: true
      },
      // motor 3
      {
        pins: {
          pwm: 4,
          dir: 27,
          enable: 26
        },
        invertPWM: true
      },
      // motor 4
      {
        pins: {
          pwm: 23,
          dir: 1,
          enable: 6
        },
        invertPWM: true
      }
    ]
  );
  motors.disable()
  motors.forward(255)
  motors.enable();
  board.wait(1000, function() {
    motors.reverse()
    board.wait(1000, function() {
      motors.stop()
    })
  });
});

```

closed 2017-07-13 by nooitaf