rwaldron/johnny-five #1364: Ping Api is not Available

salmanfarisvp, 2017-07-13:
- http://johnny-five.io/api/ping is not avilable 
- 
-  where i can connect my ultrasonic(HC-SR04) sensor to arduino. (http://johnny-five.io/examples/radar/)
    ( Servo and webserver is working fine , )

fivdi, 2017-07-13:
The HC-SR04 and additional proximity sensors are documented at http://johnny-five.io/api/proximity
There's an example circuit for a HC-SR04 on the same page at http://johnny-five.io/api/proximity/#hcsr04parallax-ping-

Did you find the link http://johnny-five.io/api/ping somewhere in the documentation? If so, can you say where please as the resource linked to doesn't exist (which you have already noticed)

fivdi, 2017-07-13:
One additional point. As special version Firmata called PingFirmata is needed for HC-SR04 proximity sensors. See http://johnny-five.io/api/proximity/#pingfirmata.

salmanfarisvp, 2017-07-13:
Thanks , i uploaded the `pingfirmata ` ,  where i nedd to connect the `HC-SR04 `pin (Echo and Trig) ..?

fivdi, 2017-07-13:
Please take a look at the first response above. There is a link pointing directly to the information you need.

salmanfarisvp, 2017-07-13:
i connected Echo and Tring on Arduino pin D7 , but did't work 

```
 // ping instance (distance detection)
  ping = new five.Ping(7);
```

fivdi, 2017-07-13:
Does it work if `ping = new five.Ping(7);` is replaced with:
```js
ping = new five.Proximity({
  controller: "HCSR04",
  pin: 7
});
```

fivdi, 2017-07-13:
The [Radar](http://johnny-five.io/examples/radar/) example appears to be a little dated. It contains the dead link to http://johnny-five.io/api/ping in section [Component Classes in this example](http://johnny-five.io/examples/radar/#component-classes-in-this-example).

salmanfarisvp, 2017-07-13:
thanks , what i need to work this

```
C:\Users\node radar.js
Option log level is not valid. Please refer to the README.
1499934123487 Available COM24,COM1
1499934123527 Connected COM24
1499934127412 Repl Initialized 
```

fivdi, 2017-07-15:
@salmanfarisvp Is the HC-SR04 working now or are there still issues? The message `Option log level is not valid. Please refer to the README.` is related to socket.io.

salmanfarisvp, 2017-07-15:
@fivdi yeah , facing same issue , can you please solve it .
we are organizing a johnny five meetup today after-noon 
https://www.meetup.com/Hackster-Hardware-Meetup-Kerala/events/241597915/

fivdi, 2017-07-15:
I can take a look at it but I'm afraid not by this afternoon.

salmanfarisvp, 2017-07-15:
thank , that's ok  :-)

fivdi, 2017-07-15:
The following program works fine for me:
```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var proximity = new five.Proximity({
    controller: "HCSR04",
    pin: 7
  });

  proximity.on("data", function() {
    console.log("cm: ", this.cm);
  });
});
```
Here's some output:
```
$ node ping.js 
1500109626850 Available /dev/ttyACM0  
1500109626872 Connected /dev/ttyACM0  
1500109630570 Repl Initialized  
>> cm:  6.306
cm:  6.306
cm:  6.186
cm:  6.186
cm:  6.186
cm:  6.306
cm:  6.306
cm:  6.306
cm:  6.168
cm:  6.168
cm:  6.168
cm:  6.254
cm:  6.254
cm:  6.254
cm:  5.808
cm:  5.808
cm:  5.808
cm:  5.808
cm:  5.808
cm:  5.808
cm:  5.79
cm:  5.79
cm:  5.79
cm:  5.79
cm:  5.79
cm:  5.79
cm:  5.79
cm:  5.79
cm:  5.704
cm:  5.704
cm:  5.704
cm:  5.808
cm:  5.808
cm:  5.808
cm:  5.808
cm:  5.808
cm:  5.808
cm:  5.79
cm:  5.79
cm:  5.79

(To exit, press ^C again or type .exit)
>> cm:  5.808
cm:  5.808
cm:  5.808
cm:  5.704
cm:  5.704
cm:  5.704
cm:  5.704
cm:  5.704
cm:  5.704

1500109631874 Board Closing.  
$ 
```
The version of PingFirmata used can be found at:  https://gist.githubusercontent.com/rwaldron/0519fcd5c48bfe43b827/raw/f17fb09b92ed04722953823d9416649ff380c35b/PingFirmata.ino

It doesn't look like there's an issue with Johnny-Five.

salmanfarisvp, 2017-07-15:
yeah , it  works fine , but problem while running the radar.js  `(http://johnny-five.io/examples/radar/)`

fivdi, 2017-07-16:
The radar example (unmodified) works fine for me. Note that radar.js was last modified on Dec 17 2013. At the time the latest version of socket.io was v0.9.16 so this was the version I installed (`npm i socket.io@0.9.16`)

Today the latest version of socket.io is v2.0.3. If you want to used socket.io v2.0.3 you'll need to update the  relevant code accordingly.

salmanfarisvp, 2017-07-17:
thanks , i Depreciated  socket.io v0.9.16 , and now i got this error . 

```
info  - socket.io started
(node:4792) DeprecationWarning: process.EventEmitter is deprecated. Use require('events') instead
```

fivdi, 2017-07-17:
That's only a warning not an error. It can be ignored. I saw it too.

salmanfarisvp, 2017-07-17:
and nothing happen

fivdi, 2017-07-17:
As mentioned above it works fine for me. What does localhost:8080 display?

salmanfarisvp, 2017-07-17:
just the image (green radar screen )

fivdi, 2017-07-17:
What's the output of the following program?
```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var proximity = new five.Ping(7);;

  proximity.on("data", function() {
    console.log("cm: ", this.cm);
  });
});
```

salmanfarisvp, 2017-07-17:
yeah , it's working fine and thanks a lot for your help , now working fine , 
i just used 
 `ping = new five.Ping(7);` 
instead of 
```
board.on("ready", function() {
  var proximity = new five.Proximity({
    controller: "HCSR04",
    pin: 7
  });
```

closed 2017-07-17 by salmanfarisvp