rwaldron/johnny-five #1365: LED Circuit Not Working

an-rohangovekar, 2017-07-17:
Hi,
I just bought the training kit and trying to make led work but I am getting this error when running  - 

```
rahul@thinkubuntu:~/Desktop/github/tessel2/jfik$ t2 run led.js
INFO Looking for your Tessel...
INFO Connected to InGen.
INFO Building project.
INFO Writing project to RAM on InGen (412.16 kB)...
INFO Deployed.
INFO Running led.js...
/tmp/remote-script/node_modules/tessel-io/lib/index.js:1
(function (exports, require, module, __filename, __dirname) { function e(e){return void 0!==e&&(__dirname=e),__dirname}function t(e){const t=+e;if(null==e||!1===e)return-1;if(!Number.isNaN(t))return t>=0&&t<O?t:-1;const r=(e=e.replace(/_|-/,""))[0].toUpperCase(),s=+e[1];if(r!==I&&r!==E&&r!==R)return-1;const i=v.BY_NAME[r].offset;return void 0===w[s+i]?-1:s+i}function r(e){e=+e;const t=v.BY_NUMBER[e>>3];return e<0||e>=O||void 0===t||void 0===w[e]?r.cache.null:(r.cache[e]||(r.cache[e]={port:t.label,index:e-t.offset}),r.cache[e])}function s(e){return"string"==typeof e&&(e=e.toUpperCase()),e===I||4===e?4:e===E||2===e?2:-1}function i(e){return"string"==typeof e&&(e=e.toUpperCase()),e===I||4===e?I:e===E||2===e?E:void 0}function n(){if(n.isActive=!0,N.length){const e=N.shift();e.port.sock.write(e.command),e.port.enqueue({size:e.size,callback(t,r){if(t)throw Error(t);e.callback(r.readUInt16LE(0)>>2),N.push(e),n()}})}}function o(e,t){return a(P-(t-e),0,P)}func

SyntaxError: Unexpected token {
    at exports.runInThisContext (vm.js:53:16)
    at Module._compile (module.js:414:25)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (/tmp/remote-script/led.js:1:69)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
```
Node Environment - 

```
rahul@thinkubuntu:~/Desktop/github/tessel2/jfik$ node -v
v4.4.4
rahul@thinkubuntu:~/Desktop/github/tessel2/jfik$ npm -v
2.15.1
```
I also tried with latest node version 6.11.1, still no luck.

rwaldron, 2017-07-18:
@an-rohangovekar can you run the following and paste the result: 

```js
t2 version
```

Thanks!

smihalka, 2017-07-23:
Hi I am having a similar problem 

t2 version
INFO Looking for your Tessel...
INFO Connected to Dwight.
INFO Tessel Environment Versions:
INFO t2-cli: 0.1.8
INFO t2-firmware: 0.0.16
INFO Node.js: 4.5.0

rwaldron, 2017-07-28:
Please update your Tessel 2 board with the command: `t2 update`. The latest version of Tessel-IO expects the latest version of t2-firmware.

closed 2017-07-28 by rwaldron