rwaldron/johnny-five #1367: Any example how to flash code onto board to work without computer

alzalabany, 2017-07-19:
For example in getting started

```
var five = require("johnny-five"),
    board = new five.Board();

board.on("ready", function() {
  // Create an Led on pin 13
  var led = new five.Led(13);

  // Strobe the pin on/off, defaults to 100ms phases
  led.strobe();
});
```

Run this and now led light,  now how can i disconnect from pc and still keel code worling ¿

dtex, 2017-07-19:
Hi @alzalabany I assume you are using an Arduino connected to your PC with a USB cable. 

The short answer is you cannot do that. The sketch that is running on the Arduino (firmata) is a client that gets all of its instructions from a host server (your pc). 

The long answer is that you can do that with a different micro-controller, or find a different way to connect to your micro-controller from your PC. Check out these options:

You can run your host server in an [onboard linux environment](http://johnny-five.io/platform-support/#relationship:embedded). I think the Tessel 2 is a great option for that.

You can connect to the micro-controller [over wifi](http://johnny-five.io/platform-support/#relationship:host-client-wifi)

You can connect to the micro-controller [over bluetooth](http://johnny-five.io/platform-support/#relationship:host-client-bluetooth)

You can also connect your PC to an Arduino using Xbee radios (which are pretty neat).

rwaldron, 2017-07-20:
Thanks @dtex!

closed 2017-07-20 by rwaldron