rwaldron/johnny-five #1368: I2C support on the ESP8266

ghtomcat, 2017-07-19:
I'm using two I2C components with an ESP8266 running StandardFirmataWifi, a LED Matrix with HTK16K33 and and ADXL345 Accelerometer. Both appear not to be working, in the case of the LED Matrix it stays dark and in the case of the Accelerometer, 0 is returned for all data. Any idea how to debug this?

dtex, 2017-07-19:
First, I'd isolate the problem to the firmata.js / StandardFirmataWifi side of things by using methods on the io object or writing a test that doesn't use Johnny-Five at all. If it works without J5, then we know were looking in the right place.

Since you are having problems with two different classes, I expect it will not work without Johnny-Five either and the problem will lie on the firmata side of things.

ghtomcat, 2017-07-20:
Good point, i will do that.

soundanalogous, 2017-07-20:
I2C definitely works with some ESP8266 boards (Adafruit feather huzzah esp8266) but appears to have issues with others (Sparkfun ESP8266 thing). Try using only one i2c device at a time to start.

Here's an example using an I2C HMC6352 digital compass with J5 on a Huzzah esp8266 over WiFi: https://github.com/soundanalogous/firmata-client-examples/blob/master/johnny-five/sensors/hmc6352.js

ghtomcat, 2017-07-24:
@dtex any pointers on how to use firmata with i2c e.g to get a simple list of i2c devices as a start?

ghtomcat, 2017-07-25:
@soundanalogous i'm actually using a Adafruit Feather Huzzah Board. I'll look at your example and test further.

soundanalogous, 2017-07-25:
Also make sure you are correctly using the EtherportClient module if you have StandardFirmataWiFi configured as a TCP server (default) or you are using the Etherport module if you have StandardFirmataWiFi configured as a TCP client.

ghtomcat, 2017-07-25:
Odd, now that i'm testing again with both the ADXL345 and the HTK16K33 display both work!

closed 2017-07-26 by ghtomcat