rwaldron/johnny-five #1374: Issue with moving the servo

ritesh2014, 2017-08-13:
My Edison board is connected to one servo (Grove). When I use the below code -
keyFrames: [ {degrees: 45}, {degrees: 90}, {degrees: 135}, {degrees: 180}, {degrees: 0}]
it works fine

but it does not moves when I use the below -
keyFrames : [ 45, 45, 45, 45, -180 ]
instead the servo makes a buzzing sound and stops after sometime.

Also, when I use -
 keyFrames: [{degrees: 90}, { value: 180, easing: "inQuad" }, { value: 90, easing: "outQuad" }, {degrees: 0}]
it works perfectly

but when I use - 
keyFrames: [90, { value: 180, easing: "inQuad" }, { value: 90, easing: "outQuad" }, -90],
the servo arm never touches 180.

It seems I have to use degrees explicitly. Just putting the values (45, 90 etc) does not works.
I am new to this framework and maybe I am doing something stupid but I really need help :(

[Note - in all above code before execution, the servo arm is at 0 degree]

dtex, 2017-08-14:
Can I see your full code?

ritesh2014, 2017-08-15:
Sure, here it is - 

var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
 var servo = new five.Servo(3);
  var animation = new five.Animation(servo);
  
///////THIS WORKS//////        
 animation.enqueue({
    cuePoints: [0, 0.25, 0.75, 1],
    keyFrames: [ {degrees: 45}, {degrees: 90}, {degrees: 135}, {degrees: 180}, {degrees: 0}],
    duration: 2000
  });

/*
///////THIS DOES NOT WORKS - the servo makes a buzzing sound and stops after sometime//////        
 animation.enqueue({
    cuePoints: [0, 0.25, 0.75, 1],
    keyFrames: [ 45, 45, 45, 45, -180 ],
    duration: 2000
  });

///////THIS WORKS//////    
 animation.enqueue({
    cuePoints: [0, 0.25, 0.75, 1],
    keyFrames: [{degrees: 90}, { value: 180, easing: "inQuad" }, { value: 90, easing: "outQuad" }, {degrees: 0}],
    duration: 2000
  });
    
///////THIS DOES NOT WORKS//////    
 animation.enqueue({
    cuePoints: [0, 0.25, 0.75, 1],
    keyFrames: [90, { value: 180, easing: "inQuad" }, { value: 90, easing: "outQuad" }, -90],
    duration: 2000
  });
*/
    
});

dtex, 2017-08-15:
Thanks. I will replicate on hardware tonight and see what's up.

dtex, 2017-08-16:
Okay, I see what's happening. Bear with me while I explain (it helps me to reason about the problem). 

With servos, passing in just a number for a keyframe means that's a step value. In other words, "move this number of steps from the previous keyframe". In your non-working examples:
````js
keyFrames: [ 45, 45, 45, 45, -180 ],
// and
keyFrames: [90, { value: 180, easing: "inQuad" }, { value: 90, easing: "outQuad" }, -90],
````
There is a number in the first keyframe so these get normalized in the Servo class to:
````js
keyFrames: [ { step: 45}, {step: 45}, {step: 45}, {step: 45}, {step: -180} ],
// and
keyFrames: [{step: 90}, { value: 180, easing: "inQuad" }, { value: 90, easing: "outQuad" }, {step: -90 }],
````
They then go through the Animation class' normalization which changes them to
````js
keyFrames: [ { value: previousVal + 45}, {value: previousVal + 45}, {value: previousVal + 45}, {value: previousVal + 45}, {value: previousVal + -180} ],
// and
keyFrames: [{value: previousVal + 90}, { value: 180, easing: "inQuad" }, { value: 90, easing: "outQuad" }, {value: previousVal + -90 }],
````
Since ```previousVal``` is undefined in the first frame you actually get:
````js
keyFrames: [ { value: NaN}, {value: NaN}, {value: NaN}, {value: NaN}, {value: NaN} ],
// and
keyFrames: [{value: NaN}, { value: 180, easing: "inQuad" }, { value: 90, easing: "outQuad" }, {value: 0 }],
````

Short answer: Don't use a step value in the first frame. It doesn't know where you are stepping from. 

Longer answer: I could change it so that if a step value is given in the first frame it will step from wherever the server was last moved to in a prior animation, the last call to servo.to, or the startAt value when it was initialized (whichever is most recent). I feel like this is an okay thing to do, but it would help my confidence if I understood your use case. Did you intend to pass a step value in the first keyframe, or did you think the number would be an explicit position? This could just be a problem with the docs.

ritesh2014, 2017-08-16:
Thanks for the detailed explanation. According to further testing, the initial frame should never be a step value. 
Even if servo is initialized with startAt:(some value) or servo.to(somevalue) is called before running the animation, giving a step value in the first frame does not works. However, giving a null as the first keyframe value works.
To answer your question -
Did you intend to pass a step value in the first keyframe, or did you think the number would be an explicit position?
I intended to pass a step value in the first keyframe and thought that it should take the initial servo value  when the servo is initialized.
Maybe the doc should be a bit clearer for a beginner (not complaining in anyway). And certainly I can contribute towards documentation if given the permission to. Thanks again.

dtex, 2017-08-16:
Having slept on it, I think the behavior you expected should be the correct behavior. Failing silently is certainly not the right thing to do so I'm going to make a change. The docs were written when animation only supported servos so that needs to be updated anyway. I'll try and a get a PR in tonight.

PR #1375 (2017-08-18): Handle number as first value of servo keyframes
When number is passed as the first keyframe of a servo animation it should be treated as a step from the most recent position.

Also adds tests for led to ensure that number values in first keyframe are still treated as explicit values.

Will close #1374 

Note that when this lands I need to update the animation wiki to better describe all devices that can be animated.
closed 2017-08-18 by rwaldron