rwaldron/johnny-five #1378: Feature: add MAVLink support

TSC21, 2017-08-18:
Any plans to add MAVLink support to the johnny-five interface?

dtex, 2017-08-19:
There is already a [Mavlink parser for node](https://www.npmjs.com/package/mavlink) that could certainly be used with any Johnny-Five project. Johnny-Five's sweet spot is at a lower level for device specific control (where a device is a pin, motor, servo, sensor, etc). Tell us more about your goals and how you would use J5 + Mavlink and how you think it would fit into J5.

rwaldron, 2017-08-21:
@dtex 👍

closed 2017-08-21 by rwaldron

TSC21, 2017-08-22:
I suppose it could serve as a link between low level device connections and Mavlink-based autopilots. This was a suggestion, nothing I'm planning to use though.