rwaldron/johnny-five #1379: serialport.node' was compiled against a different Node.js version using

korealex, 2017-08-19:
There is no version of node with 54 node 8 uses 57
Error in mounted hook: "Error: The module '/home/al/www/geospot/node_modules/johnny-five/node_modules/serialport/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`)."

[image]

rwaldron, 2017-08-21:
Looks like you need to recompile serialport for use with electron? Have a look at this: https://github.com/johnny-five-io/electron-serialport

closed 2017-08-21 by rwaldron

korealex, 2017-08-21:
Yeah, already got it working, there are other problems but found a great article about it.
[http://meow.noopkat.com/using-node-serialport-in-an-electron-app/](url).
Thanks!

rwaldron, 2017-08-21:
Ah, yes, @noopkat's guide is definitely the best.