rwaldron/johnny-five #1380: Error in compiling blink-led in Intel Galileo Gen 2

vrjgamer, 2017-08-21:
I am not able to compile the code, I am getting this error
`
root@galileo:~/smartPlug# node test.js

/home/root/node_modules/johnny-five/node_modules/chalk/index.js:2
const escapeStringRegexp = require('escape-string-regexp');
^^^^^
SyntaxError: Use of const in strict mode.
    at Module._compile (module.js:439:25)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/root/node_modules/johnny-five/lib/board.js:16:13)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
root@galileo:~/smartPlug# node --harmony test.js

/home/root/node_modules/johnny-five/node_modules/chalk/index.js:55
for (const key of Object.keys(ansiStyles)) {
               ^^
SyntaxError: Unexpected identifier
    at Module._compile (module.js:439:25)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/root/node_modules/johnny-five/lib/board.js:16:13)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
root@galileo:~/smartPlug#
`
I am using 'iot-devkit-latest-mmcblkp0.direct.bz2' for os in galileo.

rwaldron, 2017-08-23:
Thanks for filing. The issue you're hitting is not coming from Johnny-Five itself, but one of its dependencies: chalk. That module appears to no longer support Node < 6. Until now, Johnny-Five has always favored the "latest" version, but that needs to be changed.

rwaldron, 2017-08-23:
Fixed in  https://github.com/rwaldron/johnny-five/commit/e9f8beb3bea708e69402e058ea61805554bfdd80

closed 2017-08-23 by rwaldron