rwaldron/johnny-five #1382: Issue with servo.ccw(1)

ritesh2014, 2017-08-24:
I am testing a continuous servo - Tower Pro MG995 for clockwise and anticlockwise movement.

While this works - servo.cw(1) ----servo rotates in the anticlockwise direction continuously
This does not works - servo.ccw(1) ------actually the servo move a bit in the clockwise direction and then stops.
I am testing on Edison. Here is the code - 

var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {

  var servo = new five.Servo({
    pin: 5, 
    type: "continuous"
  });
     servo.cw(1); //works fine
   //servo.ccw(1);//does not works...it stops after moving a little in the opposite direction
});

Can you please help?

dtex, 2017-08-24:
Are you using external power for the servo or is it powered off the Edison?

ritesh2014, 2017-08-25:
Its powered off the Edison but the edison is connected to external power source (via DC adapter NOT through USB). The servo rotates fine in one direction but when I use CCCW(), it moves a bit in the opposite direction (say about 30 to 40 degree) and then stops.

dtex, 2017-08-25:
If invert is true then ccw becomes cw (the same value is sent to the servo). 

I'm curious if your program crashes when this happens or if otherwise it keeps on working. For example, if you set an LED to blink does it keep blinking even after the servo stops?

If blink keeps working this could be an issue with the PWM range on your servo. I checked the TowerPro site and couldn't find PWM range info, but it should be standard so this may just be an anomalous servo.

Fortunately there is an undocumented feature (that needs to change) in Johnny-Five that let's you control the PWM range. The default pwmRange is ```[450, 1850]```. 

Try this:
````
var servo = new five.Servo({
    pin: 5,
    type: "continuous",
    pwmRange: [650, 1650]
});
````

I honestly don't know what the expected behavior should be when the pwm range of the servo is exceeded, but I do know that when you don't reach the limit on either end you are not getting full speed. If narrowing the range works, you could experiment to find the max and min values for your specific servo.

ritesh2014, 2017-08-25:
When I use pwmRange: [650, 1650] , I get the error -

>> ERROR: /home/root/.node_app_slot/node_modules/johnny-five/lib/servo.js:56
          this.io.servoConfig(this.pin, opts.pwmRange[0], opts.pwmRange[1]);
                  ^

TypeError: this.io.servoConfig is not a function
    at Servo.Controllers.Standard.initialize.value (/home/root/.node_app_slot/node_modules/johnny-five/lib/servo.js:56:19)
    at new Servo (/home/root/.node_app_slot/node_modules/johnny-five/lib/servo.js:188:8)
    at Board.<anonymous> (/home/root/.node_app_slot/main.js:9:18)
    at emitNone (events.js:72:20)
    at Board.emit (events.js:166:7)
    at nextTickCallbackWith0Args (node.js:420:9)
    at process._tickDomainCallback (node.js:390:13)
    at Function.Module.runMain (module.js:443:11)
    at startup (node.js:139:18)
    at node.js:968:3

dtex, 2017-08-25:
Doh! I forgot... edison-io doesn't have that method. Let's fake it. Try: 
````servo.ccw(0.9)````

dtex, 2017-08-25:
Oh wait... edison-io/galileo-io does have that method. What version are you running?

ritesh2014, 2017-08-25:
So I updated to latest version - 
edison-io - 0.10.0
johnny-five - 0.11.6

The code pwmRange: [650, 1650] does not throws an error now but the servo does not works both for CW() and CCW().
It does not works even if I use the default pwmRange: [450, 1850] as you mentioned before.

rwaldron, 2017-08-25:
> I am testing a continuous servo - Tower Pro MG995 for clockwise and anticlockwise movement.

The Tower Pro MG995 isn't a continuous rotation servo by default, but can be modified to support that mode. Did you modify the servo?

ritesh2014, 2017-08-25:
Tower Pro MG995 servo has both continuous and standard version. I purchased the continuous one and it rotates fine in one direction. I did not modify it.

rwaldron, 2017-08-25:
Can you link to wherever you bought it from? All the documentation that I've seen shows that Tower Pro MG995 can being modified to support continuous rotation, but is not a continuous rotation servo by default. Incidentally, the behavior you're seeing is what happens when trying to control a standard servo by using a `Servo` instance with `type: "continuous"`, ie. only one direction works.

ritesh2014, 2017-08-25:
I checked couple of sites when looking for continuous servo but bought it from a local shop (same servo) since the delivery was taking long. Here are the online links -
http://www.ebay.in/itm/TowerPro-MG995-Continuous-rotation-360-Metal-Gear-Servo-Motor-9-2kg-/172695698943?hash=item2835773dff

https://robu.in/product/towerpro-mg995-continuous-rotation-360-metal-gear-servo-motor-9-2kg/

rwaldron, 2017-08-25:
Also, there is no continuous rotation variant on the Tower Pro website: http://www.towerpro.com.tw/product-category/servos-parts/x-large-servo/

dtex, 2017-08-25:
Well, that's not progress is it? Does it work if you remove the pwmRange option completely?

rwaldron, 2017-08-25:
On this page: https://robu.in/product/towerpro-mg995-continuous-rotation-360-metal-gear-servo-motor-9-2kg/

> TowerPro MG995 Metal Gear Servo Motor is a high-speed standard servo can rotate approximately 120 degrees (60 in each direction).

Which is standard servo behavior, not continuous. The other link is the same source "robu.in". 

You should pop the case open and see if there is a stopper, as shown here: http://www.instructables.com/id/Convert-Towerpro-MG995-servo-to-continuous-rotatio/#step5

ritesh2014, 2017-08-25:
ok, I will proceed to open the case now but just mentioning again that the servo moves continuously (rotates 360 degrees) when I use cw()

rwaldron, 2017-08-25:
@ritesh2014 wait! Don't open it yet! I read: 

> servo rotates in the anticlockwise direction continuously

And must of lost that while reading through the rest of the thread. Let me see if I can try to reproduce (I will open one of my Tower Pro MG995)

ritesh2014, 2017-08-25:
:) waiting...lol
I was searching for the screwdriver now!

rwaldron, 2017-08-25:
Let's try something... Switch your code to the following: 

```js
var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  var servo = new five.Servo(5);

  servo.to(180); // what happens?

  setTimeout(function() {
    servo.to(0); // what happens?
  }, 3000);

});
```

ritesh2014, 2017-08-25:
I ran the code...

servo.to(180); // what happens? ----- the servo rotates 360 degrees continuously for 3 seconds till the below code is called and then stops

 setTimeout(function() {
    servo.to(0); // what happens? --------- stops the servo 
  }, 3000);

rwaldron, 2017-08-25:
> servo.to(180); // what happens? ----- the servo rotates 360 degrees continuously for 3 seconds till the below code is called and then stops

Good.

> servo.to(0); // what happens? --------- stops the servo

Bad. 

-------------------------------------------------

Now try this: 

```js
var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  var degree = 180;
  var servo = new five.Servo(5);

  function update() {
    console.log("Setting degree: ", degree);
    console.log("Observe and record behavior!");
    servo.to(degree);

    setTimeout(function() {
      degree -= 10;
      if (degree) {
        update();
      } else {
        process.exit();
      }
    }, 3000);
  }

  update();
});
```

ritesh2014, 2017-08-25:
Well, something interesting happened...
The servo rotates 360 degrees (anticlockwise) till the countdown to 150 degrees and then after that moves in the opposite direction (clockwise) in small spikes for every 10 degree decrements till the end of countdown to 10 degrees and stops.
Here is the log - 

>> Setting degree:  180
Observe and record behavior!
Setting degree:  170
Observe and record behavior!
Setting degree:  160
Observe and record behavior!
Setting degree:  150
Observe and record behavior!
Setting degree:  140
Observe and record behavior!
Setting degree:  130
Observe and record behavior!
Setting degree:  120
Observe and record behavior!
Setting degree:  110
Observe and record behavior!
Setting degree:  100
Observe and record behavior!
Setting degree:  90
Observe and record behavior!
Setting degree:  80
Observe and record behavior!
Setting degree:  70
Observe and record behavior!
Setting degree:  60
Observe and record behavior!
Setting degree:  50
Observe and record behavior!
Setting degree:  40
Observe and record behavior!
Setting degree:  30
Observe and record behavior!
Setting degree:  20
Observe and record behavior!
Setting degree:  10
Observe and record behavior!

dtex, 2017-08-25:
Do you have another one of those servos you can test with?

ritesh2014, 2017-08-25:
Yep, I have two servos (same ones) and tested on both with the same behavior.

dtex, 2017-08-25:
Are those "small spikes" actually ten degree movements? I wonder if these were incorrectly converted standard servos. Does the packaging look legit and actually say "Continuous Rotation" on it?

ritesh2014, 2017-08-25:
I think its a 10 degree movement. Both the servos came with no legit packaging so maybe its a converted servo. But if it is then can I make it move smoothly in clockwise direction (how it moves 360 degrees anticlockwise)

dtex, 2017-08-25:
On a standard servo there is a pot that is used to provide feedback so the servo can be positioned. It will regulate voltage being sent to the motor. CR servos don't need that pot. Given the results of Rick's clever test, it really sounds like the pot is still doing its job... sorta... In one direction only, which is like super weird.

In that conversion guide Rick linked to, one of the steps is to remove the pot altogether "for use in another project". The pot must still be in yours for it to even be able to move in 10 degree increments.

ritesh2014, 2017-08-25:
hmm...so I guess I have to open the case and take a look inside...will let you know the findings...thanks guys for all your help...really appreciate!!

rwaldron, 2017-08-26:
I think we just need to set a `deadband` high/low?

Try...

```js
var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  var degree = 180;
  var servo = new five.Servo({
    pin: 5, 
    type: "continuous",
    deadband: [155, 155] // some where around here?
  });
});
```

ritesh2014, 2017-08-26:
So I tried with the deadband range also - 
board.on("ready", function() {
  var degree = 180;
  var servo = new five.Servo({
    pin: 5, 
    type: "continuous",
    deadband: [155, 155] // i changed this to different values [150, 150], [157, 157], [160, 160] etc with no luck
  });
    servo.ccw(1);//does not works but cw(1) works
});

Then I opened the servo case. Here are the pics - 
[image]
[image]

I cannot see the knob (or protrusion) which prevents the rotation. I could see a pot though - b5k-210. Do I need to do anything here...adjust or replace with resistors maybe. Will that work?

dtex, 2017-08-26:
I do think you will need to replace the pot with the resistors for it to work. Seeing the pic, I think I might have one of those servos at my lab. I'd be happy to try it out before you risk your servo, but now it looks like I may not be able to get to there until Thursday or Friday due to the storm.

ritesh2014, 2017-08-26:
I don't have the 2.2 k Ohm resistors handy. Lets see if I can get some tomorrow. If I get it then nothing's stopping me to try!!

ritesh2014, 2017-08-27:
It works!!!
But I did not use the resistors (though I managed to get the 2.2k ohms from an old tape player). 
After opening the servo case, I removed the pot. The pot was connected to the gear assembly so I figured it was sending the feedback to the circuit. 
I sent a servo.stop() signal but the servo kept moving. Then I adjusted the pot to stop the servo. I used some glue to hold that position.
After that I used the below code - 
var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
   var servo = new five.Servo({
    pin: 5, 
    type: "continuous",
  });
servo.cw();
    setTimeout(function() {
    servo.ccw(1); // what happens?
  }, 10000);
    
    setTimeout(function() {
    servo.stop(); // what happens?
  }, 20000);
});

The pot moved anticlockwise for 10 seconds and then clockwise for 10 seconds and then stopped after 10 more seconds.
Phew! and Eureka!!!

closed 2017-08-27 by ritesh2014