rwaldron/johnny-five #1383: How to receive data computed by the firmware (not coming from I/O) ?

jmchiappa, 2017-08-26:
Hi,
I've would like to add some processing in the firmata firmware loop because of some real time issues.
How can I send back these computed values to J5 ?
Do I need to (and can I) create a dedicated sensor without a physical pin ?

Which firmata message should I send to Johnny Five ? I've read that a Sysex command allows to send many data to the host but how can I declare a subscriber in J5 to an "ANALOG_MESSAGE" command?
Thanks for any advices.
jm

dtex, 2017-08-26:
You don't necessarily need to do anything in Johnny-Five. What you are looking for is here:

https://github.com/firmata/firmata.js/blob/master/lib/firmata.js#L155-L159

You could create a custom switch case for your own sysex command and from there you can emit an event that you can listen for in your code.

Alternately you could just send a short string from firmata and parse that in your code:

https://github.com/firmata/protocol/blob/master/protocol.md#string

jmchiappa, 2017-08-26:
Thank you for your quick answer.
It sounds good !

I will have a look on that, thank you.

closed 2017-08-26 by jmchiappa