rwaldron/johnny-five #1384: Tessel 2 & Expander errors

rachelnicole, 2017-08-29:
Hey Everyone, I'm pretty stumped. 

I'm using a Tessel 2 and this expander https://www.adafruit.com/product/1411 

I've got a potentiometer plugged into pin A4 on the Tessel, and the SCL & SDA on the expander board plugged into A0 & A1. 

The potentiometer works and logs data ONLY when I don't have the Expander board defined anywhere in the code. But after defining the expander, either as a virtual board or as a device controller causes the potentiometer to no longer fire on("change") events.

Furthermore, trying to use the led plugged into the expander board gives the following error:

```
✨  Desktop/CYBORG | master💩  ~> t2 run index.js
INFO Looking for your Tessel...
INFO Connected to cyborgface.
INFO Building project.
INFO Writing project to RAM on cyborgface (691.2 kB)...
INFO Deployed.
INFO Running index.js...
1498138929290 Available Tessel 2 (cyborgface)
Socket: Error occurred: Error: connect EAGAIN /var/run/tessel/port_a
1498138929655 Connected Tessel 2 (cyborgface)
1498138929697 Repl Initialized
>> /usr/lib/node/tessel-export.js:337
        throw new Error('Socket: The Port socket has closed.');
        ^

Error: Socket: The Port socket has closed.
    at Socket.Port.sock.on (/usr/lib/node/tessel-export.js:337:15)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:497:12)
```

Here is the source code for the project I'm working on. https://gist.github.com/rachelnicole/ef004cc50099f94bd9cf4817ef23caef

rachelnicole, 2017-08-29:
I also ran continuity tests through my circuits on the protoboard and the connections were fine. Idk if that is relevant info or not. Here's also some shots of my board:

[image]
[image]
[image]

I soldered the power & ground part on backwards and haven't fixed it yet lol.

rachelnicole, 2017-08-29:
I've also tried the code without the potentiometer, and just the code for the led to turn on, and I still get the same error. Also tried it with the potentiometer in Port B, and the error was the same.

rachelnicole, 2017-08-29:
UPDATE: Suz pointed out I might have forgotten some power and ground stuff, so I'm going to try that tomorrow and get back to you.

rwaldron, 2017-08-31:
Glad you got it sorted out! Sorry I didn't get a chance to get back to you sooner 😜