rwaldron/johnny-five #1386: Ultrasonic sensor error

ritesh2014, 2017-08-31:
I was using the below code to read ultrasonic sensor (HC-SR04) value - 
 new five.Proximity({
  controller: "HCSR04",
  pin: 7
but then I got the error - TypeError: this.io.pingRead is not a function. 
Then I read somewhere that ping is not supported on edison so I used the - controller: "HCSR04I2CBACKPACK" code as @rwaldron had mentioned ----

var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});
board.on("ready", function() {
var proximity = new five.Proximity({
    controller: "HCSR04I2CBACKPACK",
    freq: 500,
  });
proximity.on("data", function() {
    console.log("Proximity: ");
    console.log("  cm  : ", this.cm);
    console.log("  in  : ", this.in);
    console.log("-----------------");
});
proximity.on("change", function() {
    console.log("The obstruction has moved.");
});

Now I am getting the below error - 
ERROR: I2C: Could not read 2 Bytes from peripheral with address 0x27
ERROR: I2C: Could not read 2 Bytes from peripheral with address 0x27
Proximity: 
  cm  :  0
  in  :  0
-----------------
The obstruction has moved.
ERROR: I2C: Could not read 2 Bytes from peripheral with address 0x27
ERROR: I2C: Could not read 2 Bytes from peripheral with address 0x27

I followed the same circuit diagram as mentioned on the J5 website. Somehow it felt weird that both the Trig and Echo pins are connected together in the diagram but I followed the same thing (soldered both these points and connected it to the I2C jack on my grove shield which is mounted on the arduino board).
Am I doing anything wrong? Is the sensor fine?

ritesh2014, 2017-09-11:
Just wanted to check if anyone looked into this...

rwaldron, 2017-09-11:
There's no way to troubleshoot this without seeing your actual circuit. Please provide a close up of your circuit's wiring and components, as clear as possible.

ritesh2014, 2017-09-12:
Here is how I have soldered the pins. The echo and trigger are soldered together as indicated in the J5 page. The wiring is then connected to the I2C jack on my grove shield.
[image]

Pls let me know if you need any other details.

fivdi, 2017-09-12:
If I'm not mistaken, the HC-SR04 should be connected to a backpack and not directly to the I2C jack on the grove shield. The I2C backpack should then be connect to the I2C jack on the Edison. Something like what's shown [here](http://johnny-five.io/api/proximity/#hcsr04-and-friends-i2c-backpack-) except an Edison is used rather than an Arduino.

rwaldron, 2017-09-12:
> If I'm not mistaken, the HC-SR04 should be connected to a backpack and not directly to the I2C jack on the grove shield.

@fivdi yep—this is what I was hoping to eliminate from potential issues, but wanted to see if my hunch appeared in the circuit itself.

ritesh2014, 2017-09-14:
In examples like - http://www.instructables.com/id/Intel-Edison-Distance-Bug-HC-SR04/ when using Arduino code, the Echo and Trig pins are connected to D7 and D8 on Grove shield. Is it possible to replicate this using Node Serial port? Or do we have to use I2C back pack?

fivdi, 2017-09-14:
@ritesh2014 Some Johnny-Five IO Plugins support what's call "ping" functionality and some don't. Whether or not an IO Plugin supports ping can be seen at http://johnny-five.io/platform-support/. 

[Edison IO](http://johnny-five.io/platform-support/#intel-edison-arduino) doesn't support ping. If an IO Plugin doesn't support ping then an I2C backpack is required.

> Is it possible to replicate this using Node Serial port?

I'm afraid I don't understand how serialport could be used to help here.

ritesh2014, 2017-09-14:
Since I saw some examples (using Arduino program) where they are not using I2C backpack, I thought will it be possible to implement this in Javascript. Not necessarily using Johnny five but maybe lets say node mraa library or something else (bad choice when I mentioned serial port)...I am very new to IOT so maybe I am causing confusion here but I want to explore options before I go out and purchase an I2C backpack.

fivdi, 2017-09-14:
The Arduino program in the instructable linked to above uses `pulseIn` to measure distance. Here's the code:
```c
 digitalWrite(trigPin, LOW); 
 delayMicroseconds(2); 

 digitalWrite(trigPin, HIGH);
 delayMicroseconds(10); 
 
 digitalWrite(trigPin, LOW);
 duration = pulseIn(echoPin, HIGH);
 
 //Calculate distance based on speed of sound
 distance = duration/58.2;
```

`pulseIn` is a blocking synchronous function. There are scenarios where `pulseIn` will run for 200 milliseconds, for example, if there is nothing in front of the sensor to detect and the sensor times out after 200 milliseconds. Synchronous calls that take so long to run are not a good idea in a large number of Johnny-Five programs as they will mess up the timing for many other things like pulsing LEDs and servos.

The synchronous nature of `pulseIn` is why backpacks are preferred in Johnny-Five. The blocking code can then run on the backpack.

There are asynchronous techniques that can measure distance using a HC-SR04. For example, there's an IO Plugin for the Raspberry Pi that uses a technique called alerts to [measure distance asychronously](https://github.com/fivdi/pi-io#measure-proximity-with-a-hc-sr04-using-separate-pins-for-trigger-and-echo). I'm not familiar enough with the Edison to say if something similar is possible there.

ritesh2014, 2017-09-14:
Thanks for the detailed explanation. One last thing. Will this I2C work -
https://www.amazon.com/Adafruit-i2c-SPI-character-backpack/dp/B00OKCON84/ref=sr_1_1?ie=UTF8&qid=1505411888&sr=8-1&keywords=adafruit+i2c+%2F+spi+character+lcd+backpack

Or can you please send me a link to buy the right one? I maybe asking for too much but I cannot figure out which I2C backpack I should buy

fivdi, 2017-09-14:
You're not asking too much. Lots of this stuff isn't easy to figure out.

The "Adafruit i2c / SPI character LCD backpack" linked to above is not the correct backpack to use.

As far as I remember the term "backpack" comes from Adafruit originally. It's basically a small expansion board for an Arduino UNO. There are other terms used in similar contexts. For example, expansion boards for the Raspberry Pi are often called "Hats", expansion boards for the BeagleBone are often called "Capes".

Johnny-Five borrowed the term "backpack" from Adafruit. In the Johnny-Five world a backpack is an Arduino Nano that runs custom software to control something, for example, a HC-SR04. The backpack communicates with the Johnny-Five world via I2C. A Johnny-Five backpack isn't restricted to communicating with an Arduino UNO via I2C. It can also communicate with a Raspberry Pi, BeagleBone or Edison via I2C.

There's an example of a Johnny-Five backpack [here](https://github.com/rwaldron/johnny-five/blob/master/docs/proximity-hcsr04-i2c.md). The Arduino Nano is the backpack. The backpack controls the HC-SR04 and communicates with the Arduino UNO via I2C.

Strictly speaking a Johnny-Five backpack need not be a Arduino Nano, it can be any type of microcontroller. If you have a Arduino UNO that could be used as a backpack too.

 A sketch or firmware is needed for the backpack. The sketches for some backpacks can be found [here](https://github.com/rwaldron/johnny-five/tree/master/firmwares). The sketch for HC-SR04 backpacks can be found [here](https://github.com/ajfisher/nodebots-hcsr04/tree/master/firmware/src/controller_src/hcsr04_backpack).

Additional Information:
http://johnny-five.io/news/introducing-i2c-component-backpacks/
http://johnny-five.io/news/hcsr04-i2c-backpack-support-lands/

What you need to buy is and Arduino Nano, something like [this](https://www.amazon.com/Arduino-Elegoo-ATmega328P-without-compatible/dp/B0713XK923). As mentioned above, if you already have an Arduino UNO that could also be used.

ritesh2014, 2017-09-15:
I can't thank you enough for providing all the information. I will procure the Nano board and configure. I will keep the issue open till then (incase if I run across any problems). Or if it needs to be closed then please go ahead and close it.

ritesh2014, 2017-09-19:
So, this is what I did after getting the Nano - 

Followed this site for instructions - https://github.com/johnny-five-io/nodebots-interchange
Executed - `npm install nodebots-interchange`
Then executed - 
```
C:\Program Files\nodejs\node_modules\npm\node_modules\.bin>interchange install hc-sr04 -a nano -p COM6 -i 0x27
Installing hc-sr04 from npm
+ nodebots-hcsr04@0.5.0
added 1 package and removed 330 packages in 61.751s
Manifest file incorrect
Manifest file error
The batch file cannot be found.
```

Connected the wires as shown in the J5 site -
VCC on sensor to +5V on Nano
Ground on sensor to Ground on Nano
Trig and Echo are soldered together and connected to D8 on Nano
A5 on Nano to SCL on Edison
A4 on Nano to SDA on Edison
VIN and Ground on Nano connected to Edison

After running the program I get -
ERROR: I2C: Could not read 2 Bytes from peripheral with address 0x27
ERROR: I2C: Could not read 2 Bytes from peripheral with address 0x27
ERROR: I2C: Could not read 2 Bytes from peripheral with address 0x27
Proximity: 
  cm  :  0
  in  :  0
-----------------
The obstruction has moved.
ERROR: I2C: Could not read 2 Bytes from peripheral with address 0x27
ERROR: I2C: Could not read 2 Bytes from peripheral with address 0x27
ERROR: I2C: Could not read 2 Bytes from peripheral with address 0x27

[image]

Where am I going wrong? Is there any other way to test this?

fivdi, 2017-09-19:
It looks like the firmware wasn't flashed onto the Nano with the command `npm install nodebots-interchange` as the output indicates that there were errors:
```
C:\Program Files\nodejs\node_modules\npm\node_modules\.bin>interchange install hc-sr04 -a nano -p COM6 -i 0x27
Installing hc-sr04 from npm
+ nodebots-hcsr04@0.5.0
added 1 package and removed 330 packages in 61.751s
Manifest file incorrect
Manifest file error
The batch file cannot be found.
```
@ajfisher do you know what the problem could be here?

fivdi, 2017-09-19:
@ritesh2014 The output of `interchange install hc-sr04 -a nano -p COM6 -i 0x27` should look something like this:
```
Installing hc-sr04 from npm
+ nodebots-hcsr04@0.5.0
updated 1 package in 12.162s
connected
reset complete.
flashing, please wait...
flash complete.
Setting firmware details
Cleaning up. Installation complete.
```

ritesh2014, 2017-09-20:
Does this work on Windows? My Nano is connected to Windows on COM6 port and I am executing the commands. Or do I have to use a Linux machine?

ajfisher, 2017-09-20:
@ritesh2014 this looks like a pathing problem to me.

Please execute interchange from your project folder as it's looking for a path to your node_modules folder and not finding it is what I suspect.

Two options for you:

* cd to your project folder then run as `./node_modules/.bin/interchange ...` and that should work
* install interchange globally and then just run `interchange install... ` from your project folder and that should work as well.

ritesh2014, 2017-09-20:
@ajfisher I installed interchange globally and then executed - interchange install hc-sr04 -a nano -p COM6 -i 0x27. I got the same error - 
[image]

And now npm again does not works. Executing npm -v gives error so I guess I have to repair it again!

ajfisher, 2017-09-20:
Try it using interchange install --interactive and see what happens?

ritesh2014, 2017-09-20:
This is what I get when using interactive - 
[image]

ritesh2014, 2017-09-21:
There must be some kind of way out of here :)
@fivdi @ajfisher Okay so since this is not working...is there any alternate distance measuring sensor that I can use which does not uses this backpack thing...something which can directly plug into Edison? Basically I want to put this sensor in my vehicle to measure distance from a couple of centimeters to 5-6 meters.

ajfisher, 2017-09-21:
I'm really unsure what's going on here as it otherwise works ok.

Maybe try it this way instead:

```
interchange install git+https://github.com/ajfisher/nodebots-hcsr04 -a nano -p COM6
```

ritesh2014, 2017-09-21:
@ajfisher  This actually worked!!! Closing this issue. Thanks everyone for their help.

closed 2017-09-21 by ritesh2014

ajfisher, 2017-09-21:
Thanks @ritesh2014 can you do me a favour and just post all the details on your machine here as well so I can follow up and make sure there's nothing else the matter with the way the npm install is working?

Just need:

- OS
- Node Version
- NPM version

ritesh2014, 2017-09-22:
OS - Windows 10 Home
Node - 6.11.3
NPM - 5.4.2