rwaldron/johnny-five #1390: Using ADC (ADS1115) with a Raspberry pi

adrienbarreau, 2017-09-20:
I've the same hardware setup than described in this [link](https://www.modmypi.com/blog/raspberry-pi-plant-pot-moisture-sensor-via-analogue-signals)

My ADC is an ADS1115.

I'd like to get data from the sensor.

I've tried the following code with different pin number but I keep getting the following error
``` javascript
board.on('ready', () => {
  // Create a new generic sensor instance for
  // a sensor connected to an analog (ADC) pin
  const sensor = new five.Sensor("2");

  // When the sensor value changes, log the value
  sensor.on("change", function(value) {
    console.log(value);
  });
});
```

```
Error: Pin "2" does not support mode "analog"
    at RaspiIOCore.value (/var/app/microgrowery/node_modules/raspi-io-core/dist/index.js:539:15)
    at RaspiIOCore.pinMode (/var/app/microgrowery/node_modules/raspi-io-core/dist/index.js:505:21)
    at new Sensor (/var/app/microgrowery/node_modules/johnny-five/lib/sensor.js:81:11)
    at Board.board.on (/var/app/microgrowery/index.js:10:18)
    at emitNone (events.js:91:20)
    at Board.emit (events.js:185:7)
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickDomainCallback (internal/process/next_tick.js:128:9)
    at Module.runMain (module.js:606:11)
    at run (bootstrap_node.js:389:7)
```

Am I close ? or maybe I should use an expander ? If so, which one would it be ?

mwittig, 2017-09-20:
I think you need to initialize the Sensor to use the "digital" pin mode.

```
const sensor = new five.Digital.Sensor("2");
```

Have a loot at http://johnny-five.io/api/sensor/ which also gives more information on how to handle the data provided by the ADC

adrienbarreau, 2017-09-20:
@mwittig I updated my post to show how I setup the board

fivdi, 2017-09-20:
Johnny-Five currently doesn't support the ADS1115. The expander chips that it's supports are shown in [this table](http://johnny-five.io/api/expander/#expander-capabilities). An example of an ADC chip that it supports is the PCF8591 whose usage is shown [here](https://github.com/rwaldron/johnny-five/blob/master/docs/expander-PCF8591.md).

mwittig, 2017-09-20:
@adrienbarreau wrote:
> I tested your code but I get the following error.

Sorry, my fault. I corrected my example as described at http://johnny-five.io/api/sensor/ which I already mentioned in my previous post. However, as @fivdi pointed out the ADC chip is currently not supported. 
Despite using the PCF8591, an alternative is to use a cheap Arduino board (if 8 bit resolution is good enough).

As @rwaldron [said earlier](https://gitter.im/rwaldron/johnny-five/archives/2015/07/12) it should be relatively easy to a add support for an ADS1015 (or ADS10115 as the two are similar makes) to the expander class. It just needs an contributor to take that challenge! The necessary protocol snippets can already be found elsewhere, for example https://github.com/alphacharlie/node-ads1x15

adrienbarreau, 2017-09-20:
Awesome! I'm new to this kind of work as I come from the web development but if I manage to make it work I'll add support for ads1x15 to Johnny-Five ;)

rwaldron, 2017-09-26:
In addition to this ADS1015 or ADS1115 not being supported, in order to use an expander with a given component class, you'd need to tell the constructor about it, ie: 

```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  // Use the expander instance to create a virtual board
  var virtual = new five.Board.Virtual(
    new five.Expander("PCF8591")
  );
  
  // Tell the Sensor constructor that it will use the virtual board, 
  // instead of the default `Board` instance
  var a = new five.Sensor({
    pin: "A0",
    board: virtual
  });

  a.on("change", function() {
    console.log(this.value);
  });
});
```

mwittig, 2017-12-10:
Great. Seems like support for ADS 1115 has landed on master!
https://github.com/rwaldron/johnny-five/pull/1399

dtex, 2018-01-01:
@adrienbarreau I'm closing this since it looks like #1399 solves your issue. If I'm missing something please feel free to re-open.

closed 2018-01-01 by dtex