rwaldron/johnny-five #1392: doubts GPS

zalazarnatanael, 2017-09-20:
I was testing with the SIM808 GPS and when I use the CHANGE event it never shows the console.log

[image]

dtex, 2017-09-20:
This code is specifically for the Adafruit Ultimate GPS which uses a PA6H receiver, and you are using a breakout with a SIM808 receiver. Specifying the Adafruit breakout might not be the right thing to do.

You can try removing the breakout property and see if it just works, but I really don't know if it will. Can you send me a link to the breakout you are using so I can see the tech specs?

dtex, 2017-09-30:
Hi @zalazarnatanael just following up on this.

dtex, 2018-01-01:
Hi @zalazarnatanael I don't want to close this if you still need help. Do you have an update?

dtex, 2018-01-11:
Hi @zalazarnatanael ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for the SIM808 there.

closed 2018-01-11 by dtex