rwaldron/johnny-five #1393: J5 blocks signal/exit handlers

Overwatchh, 2017-09-24:
Hi!

I'm trying to get an event to fire just before the program stops so I can do some cleanup.

If the program is stopped by calling process.exit() somewhere, everything works as it should (exit-hook, node-cleanup and process.on('exit')) all fire.

I'd expect the same to happen with SIGINT (except that process.on('sigint') fires instead of process.on('exit')), nothing fires and I just get `sigHandler: Unhandled signal 2, terminating`
The exception to this is if REPL is enabled, then 'board.on('exit')' fires.

It looks asif j5's SIGINT handler is overriding any other handlers. I saw something about this in a different issue, but even adding `sigint: false` didn't help.

As far as I know, `sigHandler: Unhandled signal 2, terminating` means that there is no SIGINT handler, but I defined a SIGINT handler.

If I comment out the init of 'board', then all events fire as they should  (so that has to be what's causing the problem).

Here is the code: 

```
const Raspi = require('raspi-io');
const five = require('johnny-five');
const exitHook = require('exit-hook');
const nodeCleanup = require('node-cleanup');

exitHook(function () {
    console.log('exitong (exitHook)')
});

nodeCleanup(function (a, b) {
    console.log('exiting (nodeCleanup)', arguments);
});

process.on('SIGINT', function () {
    console.log('SIGINT');
});

process.on('exit', function () {
    console.log('exit');
});

const board = new five.Board({
    repl: false,
    sigint: false,
    io: new Raspi()
});

board.on('exit', function () {
    console.log("exit-board");
})

setInterval(function () {
    console.log('tick');
}, 2000);
```

node 7.10
j5 0.11.6
Rpi 3 model B

Thanks in advance for any help with this issue

mwittig, 2017-09-25:
What happpens if try without `node-cleanup`. Looks like `node-cleanup` is messing around with signal handlers. At least, it did when I tried this out on my test system (which is not Raspi, however - will try lon Raspi later).

Please try to remove  `nodeCleanup` from your test program and see what happens. I'd expect two sigint handlers triggered in this case, namely yours, and the default sigint handler installed by j5 which triggers the exit event on the board instance amd does an deffered  exit. https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L437-L447

Overwatchh, 2017-09-25:
I tried commenting out `node-cleanup` and later `exit-hook` as well, but the result is still the same.

In the code that you linked one of the conditions is to have `sigint: true` (or undefined because it gets set to true earlier in the code if you don't define it). Therefore in my case it shouldn't be running because I passed `sigint: false`

I also tried `sigint: true`, but there was no difference.

rwaldron, 2017-09-26:
What happens if you put all of that inside a "ready" handler?

```js
board.on("ready", () => {
  // all that stuff here
});
```

Overwatchh, 2017-09-26:
Wow that works! :)
But there is still somehting that doesn't make sense here.

So here is the code now:
```
const Raspi = require('raspi-io');
const five = require('johnny-five');
const exitHook = require('exit-hook');

const board = new five.Board({
    repl: false,
    sigint: true,
    io: new Raspi()
});

board.on('ready', function () {

    exitHook(function () {
        console.log('exitong (exitHook)')
    });

    process.on('SIGINT', function () {
        console.log('process.on SIGINT');
    });

    process.on('exit', function () {
        console.log('process.on exit ');
    });

    board.on('exit', function () {
        console.log("exit-board");
    });

});

setInterval(function () {
    console.log('tick');
}, 2000);
```

And here is the output:
[image]
The first is with `sigint: false`, the second with `sigint: true`.

With both of these the exitHook now triggers as well as `process.on('exit')`. Since I closed it with ctrl-c, I'd expect `process.on('SIGINT')` to trigger in stead.
It looks like j5 catches the SIGINT event and fires `process.exit()` regardless of the sigint setting, but does not close the connection with the board (and trigger `board.on('exit')`) if `sigint: false` is passed.

Is this intentional? It seems a bit unintuitive. You'd think that passing `sigint: false` would make j5 ignore the SIGINT event completely.

rwaldron, 2017-09-26:
I'm not really sure I understand what's going on, but only because I don't know what the other modules are doing or are supposed to do (at the moment I don't have any spare time to dig into `exitHook`). Based on the screen shot and the code above, it looks like everything is correct and behaving exactly as it should. The first run, `sigint: false`, doesn't reach `this.warn("Board", "Closing.");` and then doesn't print `[timestamp] Board Closing`; the things printed coming from your console.logs here: 

```js
    exitHook(function () {
        console.log('exitong (exitHook)')
    });
    // ...
    process.on('exit', function () {
        console.log('process.on exit ');
    });
```

I apologize if I've misunderstood :\

Overwatchh, 2017-09-26:
That was it. It was `exit-hook` that was overriding `process.on('SIGINT')`, not j5. It all makes sense now. :)

I'm sorry to have wasted your time. This was just me using things wrong.

Thanks for the help! :)

rwaldron, 2017-09-26:
> I'm sorry to have wasted your time

No such thing! I'm glad we could get to the bottom of this :)

closed 2017-09-26 by rwaldron