rwaldron/johnny-five #1394: Question: Host Johnny-five server with express js and socket.io

ralphchristianeclipse, 2017-09-30:
Thanks for this wonderful port for javascript to control arduino. :)

So my question is how can we host the johnny-five server

My idea is installing a wifi module on the arduino then connect to the internet once connected.

The board will be controlled by the server but based on the documentation the package will connect to a serial port. I've also seen a package where you can use an ethernetport to connect to the arduino instead.

Any ideas on this?

ralphchristianeclipse, 2017-09-30:
and also ill host it using a dockerized contsiner

dtex, 2018-01-01:
Having a Johnny-Five host server respond to commands sent via socket.io is such a common use-case (and common question here in issues) that I think someone should create a repo with the basic code in place and instructions on how to get it running. It would be great to have something we can point people to.

RickBr0wn, 2018-01-03:
Hi... This is my first ever Issue Response I've ever done, so please be gentle, the internet can be such a harsh place, and I'm not sure if I'm answering this correctly. But here goes.

@ralphchristianeclipse I agree, johnny five is wonderful, I am only just starting out with it, but am already completely hooked!!
@dtex I am working on this repo https://github.com/RickBr0wn/DS18b20-DEMO and I think it does/demonstrates what you suggested?

excalliburbd, 2018-03-08:
I've actually done something like this a long while ago with Meteor.js and an Arduino board. https://github.com/excalliburbd/demoInterface

I would love to work on a tutorial like this. Maybe using Proteus since I don't have any wifi shields. I'll get back here if I can whip up something decent.

vettorazi, 2018-05-20:
hey! in my github there's a project using socket.io, express, johnny-five.
https://github.com/vettorazi/fila-io
but if you want something more simple, i have this code, that is a sopcket.io with j5 in the DOM:
https://gist.github.com/vettorazi/8d695f2b7697e37210481e3308c1987d

developer239, 2018-07-17:
@excalliburbd Could you please list components that you used to make that work? I also wanted to create simple LCD screen that would be wireless (with its own power source) and I could control it from anywhere with web browser.

excalliburbd, 2018-07-20:
@developer239 It was a pretty basic setup-- same as the [Arduino tutorial for LCD](https://www.arduino.cc/en/Tutorial/LiquidCrystalDisplay). If you want wireless you could use a Raspberry Pi for that.

closed 2019-07-22 by stale[bot]