rwaldron/johnny-five #1395: Bluetooth connection issue HC-06, Error opening COM port

seem-less, 2017-10-02:
As seen here 
[https://stackoverflow.com/questions/46519776/bluetooth-connection-issue-johnny-five-hc-06](url) 

I am trying to communicate with my Arduino Uno through a HC-06 Bluetooth module using Johnny-Five. I followed these instructions to change the baud rate of the module, and then re-uploaded the StandardFirmata sketch.

After running:

`node bluetooth.js`

Where "bluetooth.js" is the script below:

```
var five = require("johnny-five");
var board = new five.Board({
   port: "COM6"
});

board.on("ready", function(){
   var motor = new five.Motor(3);
   motor.start(50);
});
```
I get the following error:

```
1506917321334 Connected COM6
1506917321338 Error Opening COM6: File not found
events.js:165
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
```
The module is paired to the computer and I am getting 2 COM ports shown in Windows Bluetooth Settings. One is COM6, the other is COM7, which are labelled as outgoing and incoming respectively. I have tried both ports in the Johnny-five script.

It does run if the Arduino is connected by USB cable (COM3).

The module is blinking red when paired to the PC.

rwaldron, 2017-10-04:
@monteslu do you have any experience or guidance to share?

monteslu, 2017-10-04:
haven't tried this specific module, but just ordered one.  Will test as soon as I get it :)

seem-less, 2017-10-08:
After some more experimentation and realizing that the HC-06 LED was flashing even after pairing, this meant that the module was not actually getting connected.  So with my USB to TTL I changed the BAUD rate to 9600 and tried connecting to it directly with TeraTerm/Arduino.  TeraTerm says that it could not open the COM port and Arduino said the same thing.

After reading this [thread](https://answers.microsoft.com/en-us/windows/forum/windows_10-hardware/with-windows-10-cant-open-any-bluetooth-virtual/73b985bb-43d2-4dd3-968c-6ddf4abfbb46?auth=1) I found this:

> FINALLY, AN ANSWER.
> I have been told Windows 10 has an all-new Bluetooth stack.
> This new stack only supports SPP 1.2.  We needed 1.1 supported.
> Migrating to RFCOMM.

So I assume that the module needs SPP (Serial Port Profile) 1.1 that is not provided in Windows 10.

Because of this I found [Windows Bluetooth Serial Terminal](https://www.microsoft.com/en-us/store/p/bluetooth-serial-terminal/9wzdncrdfst8) which claims to:

> use the RFCOMM protocol to communicate with serial bluetooth devices.

I connected the device using [this](http://www.collideabq.com/2016/04/28/bluetooth-serial-connection-with-windows-10/) tutorial.

So after installing this I was able to connect with the module and tested it with [this](http://www.instructables.com/id/Add-bluetooth-to-your-Arduino-project-ArduinoHC-06/) simple echo sketch.

So now that I have a solid connection to the module and can get a response from it, I would like to use Johnny-five to control it.

So I connect to the module using Windows Bluetooth Serial Terminal, making sure that the module is indicating that it is connected with it's solid LED.  This time I get a new error message but still using COM6 as the port in the johnny-five script.

```
1507466546264 Connected COM6
1507466548324 Error Opening COM6: Unknown error code 1168
events.js:165
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
```

Could this be because RFCOMM protocol hasn't been integrated into the johnny-five library? It seems like there needs to be something from the [bluetooth-serial-port](https://www.npmjs.com/package/bluetooth-serial-port) or [windows.devices.bluetooth.rfcomm](https://www.npmjs.com/package/windows.devices.bluetooth.rfcomm) npm modules that need to be included in the Johnny-five library, but that is my guess.

closed 2019-05-22 by stale[bot]