rwaldron/johnny-five #1396: raspberry nodered Cannot read property 'type' of null  at Function.Pins.normalize (/root/.node-red/node_modules/johnny-five/lib/board.pins.js:91:24)

Waterbrain, 2017-10-03:
**Hello: 

Here is the content of /lib/systemd/system/nodered.service** 

_#systemd service file to start Node-RED

[Unit]
Description=Node-RED graphical event wiring tool.
Wants=network.target
Documentation=http://nodered.org/docs/hardware/raspberrypi.html

[Service]
Type=simple
# Run as normal pi user - feel free to change...
User=root
Group=root
WorkingDirectory=/root
Nice=5
Environment="NODE_OPTIONS=--max_old_space_size=256"
# uncomment and edit next line if you need an http proxy
#Environment="HTTP_PROXY=my.httpproxy.server.address"
# uncomment the next line for a more verbose log output
Environment="NODE_RED_OPTIONS=-v"
#ExecStart=/usr/bin/env node $NODE_OPTIONS red.js $NODE_RED_OPTIONS
ExecStart=/usr/bin/env node-red-pi $NODE_OPTIONS $NODE_RED_OPTIONS
# Use SIGINT to stop
KillSignal=SIGINT
# Auto restart on crash
Restart=on-failure
# Tag things in the log
SyslogIdentifier=Node-RED
#StandardOutput=syslog

[Install]
WantedBy=multi-user.target_

**Here is settings.js in the part of definition global:**

    _functionGlobalContext: {
         os:require('os'),
        // octalbonescript:require('octalbonescript'),
         raspiio:require("raspi-io"),
         johnnyfive:require("johnny-five"),
         j5board:require("johnny-five").Board({repl:false})
    },_

Here is my code:

__const Raspi = context.global.raspiio;
const five = context.global.johnnyfive;
const board = new five.Board({io: new context.global.raspiio()});
var reles = {};
var servos = {};

board.on("ready", function() {
  reles.rele1 = new five.Relay('GPIO1');
  reles.rele2 = new five.Relay('GPIO2');
  reles.rele3 = new five.Relay('GPIO3');
  reles.rele4 = new five.Relay('GPIO4');
  reles.rele5 = new five.Relay('GPIO5');
  reles.rele6 = new five.Relay('GPIO6');
  reles.rele7 = new five.Relay('GPIO7');
  reles.rele8 = new five.Relay('GPIO8');
  servos.camServo = new five.Servo('GPIO0');  
});

if (reles.rele1) {reles.rele1.toggle()}

return msg;__

And the error:

_Welcome to Node-RED 
=================== 
3 Oct 11:25:24 - [info] Node-RED version: v0.17.5 
3 Oct 11:25:24 - [info] Node.js  version: v6.11.3 
3 Oct 11:25:24 - [info] Linux 4.9.40-v7+ arm LE 
3 Oct 11:25:26 - [info] Loading palette nodes 
3 Oct 11:25:38 - [info] Settings file  : /root/.node-red/settings.js 
3 Oct 11:25:38 - [info] User directory : /root/.node-red 
3 Oct 11:25:38 - [info] Flows file     : /root/.node-red/flows_plantilla.json 
3 Oct 11:25:38 - [info] Server now running at http://127.0.0.1:1880/ 
3 Oct 11:25:38 - [info] Starting flows 
1507029938971 Available RaspberryPi-IO 
3 Oct 11:25:39 - [info] Started flows 
1507029939370 Connected RaspberryPi-IO 
3 Oct 11:25:39 - [info] [nodebot:RaspberryPi-IO] io ready 
3 Oct 12:37:35 - [info] Stopping flows 
3 Oct 12:37:35 - [info] [nodebot:RaspberryPi-IO] closing nodebot 
3 Oct 12:37:35 - [info] [nodebot:RaspberryPi-IO] port closed 
3 Oct 12:37:35 - [info] Stopped flows 
3 Oct 12:37:35 - [info] Starting flows 
1507034255700 Available RaspberryPi-IO 
3 Oct 12:37:35 - [info] Started flows 
1507034255854 Connected RaspberryPi-IO 
3 Oct 12:37:35 - [info] [nodebot:RaspberryPi-IO] io ready 
1507034260172 Available RaspberryPi-IO 
1507034260200 Connected RaspberryPi-IO 
1507034260211 Repl Initialized 
>> 3 Oct 12:37:40 - [red] Uncaught Exception: 
3 Oct 12:37:40 - TypeError: Cannot read property 'type' of null 
at Function.Pins.normalize (/root/.node-red/node_modules/johnny-five/lib/board.pins.js:91:24) 
at Relay.Board.Component (/root/.node-red/node_modules/johnny-five/lib/board.js:855:23) 
at new Relay (/root/.node-red/node_modules/johnny-five/lib/relay.js:14:19) 
at Board.<anonymous> (evalmachine.<anonymous>:9:17) 
at emitNone (events.js:91:20) 
at Board.emit (events.js:185:7) 
at _combinedTickCallback (internal/process/next_tick.js:73:7) 
at process._tickDomainCallback (internal/process/next_tick.js:128:9) 
nodered.service: main process exited, code=exited, status=1/FAILURE 
Unit nodered.service entered failed state. 
nodered.service holdoff time over, scheduling restart. 
Stopping Node-RED graphical event wiring tool...._ 

Can you help my?

rwaldron, 2017-10-04:
I believe you're using the wrong pin names? According to this: https://github.com/nebrius/raspi-io/wiki/Pin-Information: 

- `GPIO1` does not exist
- `GPIO2` does not exist
- `GPIO3` does not exist
- `GPIO0` does not exist

Waterbrain, 2017-10-05:
**Thanks for your answer, Apparently not of that, the output of "gpio readall" gives the following:**
 +-----+-----+---------+------+---+----Pi 3---+---+------+---------+-----+-----+
 |.BCM.|.wPi...|....Name.|.Mode.|.V...|.Physical....|.V...|.Mode|.Name....|.wPi...|.BCM.|
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |.........| ........|........3.3v.|...........|......|.....1.||.2....|.......|..........|.5v..........|..........|.........|
 |......2.|......8.|.....SDA.1.|..ALT0.|...1.|.....3.||.4....|.......|..........|.5v..........|..........|.........|
 |......3.|......9.|.....SCL.1.|..ALT0.|....1.|....5.||.6....|.......|..........|.0v..........|..........|.........|
 |......4.|......7.|..GPIO. 7.|.......IN.|....0.|....7.||.8....|.1....|.ALT0.|.TxD.......|.15......|.14....|
 |.........|.........|..........0v.|...........|.......|....9.||.10...|.1....|.ALT0.|.RxD.......|.16......|.15....|
 |....17.|......0.|...GPIO.0.|...OUT.|....0.|...11.||.12...|.0....|.OUT..|.GPIO. 1.|.1........|.18....|
 |....27.|......2.|...GPIO.2.|...OUT.|....0.|...13.||.14...|......|..........|.0v.........|...........|........|
 |....22.|......3.|...GPIO.3.|......IN.|....0.|...15.||.16...|.0....|.OUT..|.GPIO.4.|.4........|.23...|
 |.........|.........|.......3.3v.|..........|........|...17.||.18...|.0....|.OUT..|.GPIO.5.|.5........|.24...|
 |....10.|.....12.|.....MOSI.|...OUT.|....0.|...19.||.20...|......|...........|.0v........|...........|........|
 |.....9.|.....13.|.....MISO.|...OUT.|....0.|...21.||.22....|.0...|.OUT...|.GPIO.6.|.6.......|.25...|
 |....11.|.....14.|.....SCLK.|...OUT.|....0.|...23.||.24...|.0...|.OUT...|.CE0......|.10.....|.8.....|
 |........|..........|..........0v.|...........|.......|...25.||.26...|.0...|.OUT...|.CE1.......|.11.....|.7.....|
 |.....0.|.....30.|....SDA.0.|.......IN.|....1.|...27.||.28...|.1...|.IN.......|.SCL.0....|.31....|.1.....|
 |.....5.|.....21.|.GPIO.21.|...OUT.|....0.|...29.||.30...|......|............|.0v.........|.........|.......|
 |.....6.|.....22.|.GPIO.22.|...OUT.|...0.|...31.||.32....|.0...|.OUT...|.GPIO.26|.26....|.12...|
 |....13.|....23.|.GPIO.23.|...OUT.|...0.|...33.||.34....|......|...........|.0v..........|.........|.......|
 |....19.|....24.|.GPIO.24.|...OUT.|...0.|...35.||.36....|.0...|.OUT...|.GPIO.27|.27....|.16...|
 |....26.|....25.|.GPIO.25.|...OUT.|...0.|...37.||.38....|.0...|.OUT...|.GPIO.28|.28....|.20...|
 |........|..........|..........0v.|............|......|...39.||.40....|.0...|.OUT...|.GPIO.29|.29....|.21...|
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |.BCM.|.wPi...|....Name.|.Mode.|.V...|.Physical....|.V...|.Mode|.Name....|.wPi...|.BCM.|
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+

**anyway following your link and leaving my code like this:**

const Raspi = context.global.raspiio;
const five = context.global.johnnyfive;
const board = new five.Board({io: new context.global.raspiio()});
var reles = {};

board.on("ready", function() {
  reles.rele1 = new five.Relay('GPIO17');
});

if (reles.rele1) {reles.rele1.toggle()}

return msg;

**we obtain:**

5 Oct 10:27:47 - [red] Uncaught Exception: 
5 Oct 10:27:47 - TypeError: Cannot read property 'type' of null 
at Function.Pins.normalize (/root/.node-red/node_modules/johnny-five/lib/board.pins.js:91:24) 
at Relay.Board.Component (/root/.node-red/node_modules/johnny-five/lib/board.js:855:23) 
at new Relay (/root/.node-red/node_modules/johnny-five/lib/relay.js:14:19) 
at Board.<anonymous> (evalmachine.<anonymous>:9:17) 
at emitNone (events.js:91:20) 
at Board.emit (events.js:185:7) 
at _combinedTickCallback (internal/process/next_tick.js:73:7) 
at process._tickDomainCallback (internal/process/next_tick.js:128:9) 
nodered.service: main process exited, code=exited, status=1/FAILURE 

**Dive into board.pins.js and board.js, I do not see any reference to a raspberrypi board.
Is this significant?**

Waterbrain, 2017-10-05:
**That way looks better**

[image]

**Using jonnyfive's gpio works*

[image]

[image]

rwaldron, 2017-10-09:
> Dive into board.pins.js and board.js (at Board.<anonymous>), I do not see any reference to a raspberrypi board.
> Is this significant?

[Raspi-io](https://github.com/nebrius/raspi-io) would provide its own pin name mappings 

...

There are some issues with this code: 

```js
const Raspi = context.global.raspiio;
const five = context.global.johnnyfive;
const board = new five.Board({io: new context.global.raspiio()});
var reles = {};

board.on("ready", function() {
reles.rele1 = new five.Relay('GPIO17');
});

if (reles.rele1) {reles.rele1.toggle()}

return msg;
```

Specifically that `board`'s "ready" event is fired asynchronously: 

```js
board.on("ready", function() {
reles.rele1 = new five.Relay('GPIO17');
});
```

Which means that this: 

```js
if (reles.rele1) {reles.rele1.toggle()}
```

...is executed before "ready" is fired, so `if (reles.rele1)` will never evaluate to `true`. 

But none of that is related to your actual issue. Instead of using the `"GPIO17"` pin name, can you try using `"P1-11"`?

Also, let's summon @nebrius

Waterbrain, 2017-10-09:
Thank you

```javascript
const Raspi = context.global.raspiio;
const five = context.global.johnnyfive;
const board = new five.Board({io: new context.global.raspiio()});
var reles = {};

board.on("ready", function() {
  reles.rele1 = new five.Relay("P1-11");

});

if (reles.rele1) {reles.rele1.toggle()}

return msg;
```
But the issue is not solved.
```
TypeError: Cannot read property 'type' of null 
at Function.Pins.normalize (/root/.node-red/node_modules/johnny-five/lib/board.pins.js:91:24) 
at Relay.Board.Component (/root/.node-red/node_modules/johnny-five/lib/board.js:855:23) 
at new Relay (/root/.node-red/node_modules/johnny-five/lib/relay.js:14:19) 
at Board.<anonymous> (evalmachine.<anonymous>:9:17) 
at emitNone (events.js:91:20) 
at Board.emit (events.js:185:7) 
at _combinedTickCallback (internal/process/next_tick.js:73:7) 
at process._tickDomainCallback (internal/process/next_tick.js:128:9) 
nodered.service: main process exited, code=exited, status=1/FAILURE 
Unit nodered.service entered failed state. 
nodered.service holdoff time over, scheduling restart. 
```

nebrius, 2017-10-09:
Looking at https://github.com/rwaldron/johnny-five/issues/1396#issuecomment-334440766, I _suspect_ this is a pin naming issue. I'm not really familiar with Node Red, other than users having issues with it, so this is just a guess.

You have the pin set to `21` and the name set to `GPIO21`. This doesn't seem right to me, because the pin number for `GPIO21` is `2`, not `21`. I don't know what Node Red is doing with these two values, but my _guess_ is that it's taking the name just for display purposes inside the Node Red GUI, i.e. it doesn't pass them to Raspi IO. That said, pin `21` does map to `GPIO5` on all but the oldest of Raspberry Pis

Which Raspberry Pi are you running on?

Waterbrain, 2017-10-10:
Raspberry Pi 3 Model B Board 1GB

Waterbrain, 2017-10-10:
**This is a proof of concept**

[image]

**GPIO´s configuration with wPi pins:**

[image]

[image]

[image]

[image]

**Flow:**

```
[{"id":"398584e7.8b306c","type":"subflow","name":"Stepper Motor","info":"","in":[{"x":75,"y":64,"wires":[{"id":"e489e96f.35dd28"}]}],"out":[{"x":673,"y":33,"wires":[{"id":"c3df9a5f.d68248","port":0}]},{"x":673,"y":81,"wires":[{"id":"c3df9a5f.d68248","port":1}]},{"x":672,"y":132,"wires":[{"id":"c3df9a5f.d68248","port":2}]},{"x":671,"y":182,"wires":[{"id":"c3df9a5f.d68248","port":3}]},{"x":675,"y":230,"wires":[{"id":"c3df9a5f.d68248","port":4},{"id":"7c2a439d.e8a0ac","port":0},{"id":"cc877466.3b3e48","port":0}]}]},{"id":"7c2a439d.e8a0ac","type":"function","z":"398584e7.8b306c","name":"Toggle Direction","func":"var direction = flow.get('direction')||0;\n\nif (direction === -1)\n{\n    direction = 1;\n}\nelse {\n    direction = -1;\n}\nflow.set('direction',direction);\nmsg.payload = direction;\nreturn msg;","outputs":1,"noerr":0,"x":324,"y":211,"wires":[[]]},{"id":"e489e96f.35dd28","type":"switch","z":"398584e7.8b306c","name":"Step or Direction","property":"topic","propertyType":"msg","rules":[{"t":"eq","v":"Step","vt":"str"},{"t":"eq","v":"Direction","vt":"str"},{"t":"eq","v":"4or8steps","vt":"str"}],"checkall":"true","outputs":3,"x":166,"y":121,"wires":[["e7be9cc2.5f9f3"],["7c2a439d.e8a0ac"],["cc877466.3b3e48"]]},{"id":"78aea1dd.dac6c","type":"comment","z":"398584e7.8b306c","name":"Flow Variables","info":"* Direction\n* Sequence\n* SequenceIndex","x":125,"y":20,"wires":[]},{"id":"c3df9a5f.d68248","type":"function","z":"398584e7.8b306c","name":"Sequence Generator","func":"var seqindex = context.get('seqindex')||0;\nvar sequence = [];\nvar sequence8p = [ [1,0,0,0],[1,1,0,0],[0,1,0,0],[0,1,1,0],[0,0,1,0],[0,0,1,1],[0,0,0,1],[1,0,0,1]];\nvar sequence4p = [ [1,1,0,0],[0,1,1,0],[0,0,1,1],[1,0,0,1]];\nvar direction = flow.get('direction')||1;\nvar steps = flow.get('steps')||8;\n\nif (steps == 4) \n    {sequence = sequence4p;}\nelse \n    {sequence = sequence8p;}\nseqindex = (seqindex % (sequence.length));\nif (seqindex == -1) {seqindex = sequence.length-1;}\n\nmsg.payload = sequence[seqindex];\nmsg.seqindex = seqindex;\nmsg.direction = direction;\nmsg.steps = steps;\n\ncontext.set('seqindex',seqindex+direction);\n\nnode.status({fill:\"green\",shape:\"dot\",text:\"Step\"});\nnode.status({});\n\nreturn GenerateOutput(seqindex);\n\nfunction GenerateOutput(index) {\n    return [ NewMessage(sequence[index][0]), NewMessage(sequence[index][1]),NewMessage(sequence[index][2]),NewMessage(sequence[index][3]), msg  ];\n}\n\nfunction NewMessage(val) {\n    return { payload:val };\n}","outputs":"5","noerr":0,"x":496,"y":125,"wires":[[],[],[],[],[]]},{"id":"e7be9cc2.5f9f3","type":"function","z":"398584e7.8b306c","name":"Multistep","func":"var outputMsgs = [];\nvar steps = msg.payload;\nfor (var count = 0; count < steps; count++) { \n    outputMsgs.push({payload:1});\n}\nreturn [ outputMsgs ];","outputs":1,"noerr":0,"x":309,"y":50,"wires":[["c3df9a5f.d68248"]]},{"id":"48e593c0.5270dc","type":"delay","z":"398584e7.8b306c","name":"","pauseType":"rate","timeout":"5","timeoutUnits":"seconds","rate":"2","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":488,"y":46,"wires":[[]]},{"id":"cc877466.3b3e48","type":"function","z":"398584e7.8b306c","name":"4or8steps","func":"var steps = flow.get('steps')||8;\n\nif (msg.payload === 8)\n{\n    steps = 8;\n}\nelse if (msg.payload === 4)\n{\n    steps = 4;\n}\nelse {\n    steps = 8;\n}\nflow.set('steps',steps);\nmsg.payload = steps;\nreturn msg;","outputs":1,"noerr":0,"x":297,"y":268,"wires":[[]]},{"id":"499d12c3.6512fc","type":"inject","z":"8b5dffb7.f37ac","name":"Step","topic":"Step","payload":"1","payloadType":"num","repeat":"","crontab":"","once":false,"x":98,"y":80,"wires":[["339fa3b6.afd48c"]]},{"id":"90c51186.cf9bf","type":"inject","z":"8b5dffb7.f37ac","name":"Direction","topic":"Direction","payload":"1","payloadType":"num","repeat":"","crontab":"","once":false,"x":108,"y":128,"wires":[["339fa3b6.afd48c"]]},{"id":"339fa3b6.afd48c","type":"subflow:398584e7.8b306c","z":"8b5dffb7.f37ac","name":"","x":320,"y":144,"wires":[["3bd71fa5.bd4f3"],["ce3dbf82.da5a4"],["f6fa69d8.76f5b8"],["89214f40.b6b8a"],[]]},{"id":"fb3cbbd2.0f87b8","type":"inject","z":"8b5dffb7.f37ac","name":"Multistep","topic":"Step","payload":"4096","payloadType":"num","repeat":"","crontab":"","once":false,"x":108,"y":32,"wires":[["339fa3b6.afd48c"]]},{"id":"3bd71fa5.bd4f3","type":"gpio out","z":"8b5dffb7.f37ac","name":"A1","state":"OUTPUT","pin":"3","i2cDelay":"0","i2cAddress":"","i2cRegister":"","outputs":0,"board":"db2a2d70.1cc2f","x":530,"y":48,"wires":[]},{"id":"ce3dbf82.da5a4","type":"gpio out","z":"8b5dffb7.f37ac","name":"A2","state":"OUTPUT","pin":"4","i2cDelay":"0","i2cAddress":"","i2cRegister":"","outputs":0,"board":"db2a2d70.1cc2f","x":530,"y":112,"wires":[]},{"id":"f6fa69d8.76f5b8","type":"gpio out","z":"8b5dffb7.f37ac","name":"A3","state":"OUTPUT","pin":"0","i2cDelay":"0","i2cAddress":"","i2cRegister":"","outputs":0,"board":"db2a2d70.1cc2f","x":530,"y":176,"wires":[]},{"id":"89214f40.b6b8a","type":"gpio out","z":"8b5dffb7.f37ac","name":"A4","state":"OUTPUT","pin":"1","i2cDelay":"0","i2cAddress":"","i2cRegister":"","outputs":0,"board":"db2a2d70.1cc2f","x":530,"y":240,"wires":[]},{"id":"ac8e19cc.f24718","type":"inject","z":"8b5dffb7.f37ac","name":"off leds","topic":"Direction","payload":"0","payloadType":"num","repeat":"","crontab":""," [... cut ...] epeat":"","crontab":"","once":false,"x":98,"y":224,"wires":[["339fa3b6.afd48c"]]},{"id":"db2a2d70.1cc2f","type":"nodebot","z":"","name":"RaspberryPi-IO","username":"","password":"","boardType":"raspi-io","serialportName":"","connectionType":"local","mqttServer":"","socketServer":"","pubTopic":"","subTopic":"","tcpHost":"","tcpPort":"","sparkId":"","sparkToken":"","beanId":"","impId":"","meshbluServer":"https://meshblu.octoblu.com","uuid":"","token":"","sendUuid":""}]
```

**It works perfectly.**

nebrius, 2017-10-11:
Hmm. The RPi 3 definitely supports that pin.

@rwaldron, looking at the stack trace, it's crashing at https://github.com/rwaldron/johnny-five/blob/master/lib/board.pins.js#L91 because `pins` itself is null, not because of an entry in an array called `pins` is null. Is `new five.Board#pins` populated from `five.board#io.pins`?

Waterbrain, 2017-10-13:
Hello nebrius:
Sorry, but I do not understand what you're asking.
What check you suggest I make?

nebrius, 2017-10-16:
@Waterbrain sorry, it seems I wasn't clear. I had something for @rwaldron to check.

rwaldron, 2017-10-17:
> Thank you
>
> ```javascript
> const Raspi = context.global.raspiio;
> const five = context.global.johnnyfive;
> const board = new five.Board({io: new context.global.raspiio()});
> var reles = {}; [...]

Just to be clear, even if changing the pin name fixed the thrown exception, this line: 

> if (reles.rele1) {reles.rele1.toggle()}

Would never actually execute. 

@nebrius I see what you're seeing now, and to answer your question: 

> Is new five.Board#pins populated from five.board#io.pins?

Yes, absolutely—this is critical to the IO-Plugin architecture. Walking through this step-by-step: 

1. `Relay` calls [`Board.Component`](https://github.com/rwaldron/johnny-five/blob/cae125aecb7f156ec1aa7dcf2d4cbc47c868c763/lib/relay.js#L14-L16) with an instance of [`Board.Options(...)`](https://github.com/rwaldron/johnny-five/blob/cae125aecb7f156ec1aa7dcf2d4cbc47c868c763/lib/relay.js#L14-L16
), which looks like: `Options { pin: "P1-11" }` 
2. [`Board.Component`](https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L806-L807) calls `Board.mount(opts);` which returns the cached instance of `five.Board`, which itself has the instance of `Raspiio()` stored on its [`board.io` property](https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L807)
3. [`Board.Component`](https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L815-L817) then stores `opts.pin` in an array assigned to a variable called `originalPins`
4. [`Board.Component` then calls `Board.Component.initialization(componentOpts);`](https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L852) where `componentOpts` is an empty object. [`Board.Component.initialization(componentOpts);`](https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L983-L990) returns an object that looks like `{ requestPin: true, normalizePin: true }`
5. With `componentOpts.normalizePin` set to `true`, [the next condition evaluates to `true`](https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L854-L856), and [`opts = Board.Pins.normalize(opts, this.board);`](https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L855) is executed.
6. Inside the call to [`Board.Pins.normalize`](https://github.com/rwaldron/johnny-five/blob/master/lib/board.pins.js#L90-L91), the program fails at the first line: https://github.com/rwaldron/johnny-five/blob/master/lib/board.pins.js#L91 (as @nebrius has determined above

@Waterbrain I need you run the following program and report the output: 
```js
const Raspi = context.global.raspiio;
const five = context.global.johnnyfive;

const io = new Raspi();
console.log(io.pins);

const board = new five.Board({ io });

board.on("ready", function() {
  console.log(board.pins);
});
```

Waterbrain, 2017-10-17:
**Sorry I'm late. here it is:**
```
1508258664720 Available RaspberryPi-IO 
1508258664744 Connected RaspberryPi-IO 
1508258664756 Repl Initialized 
>> Pins { 
'0': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'1': 
{ supportedModes: [ 0, 1, 3, 4 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'2': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'3': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'4': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'5': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'6': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'7': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'8': 
{ supportedModes: [], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'9': 
{ supportedModes: [], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'10': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'11': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'12': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'13': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'14': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'15': 
{ supportedModes: [], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'16': 
{ supportedModes: [], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'17': 
{ supportedModes: [], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'18': 
{ supportedModes: [], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'19': 
{ supportedModes: [], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'20': 
{ supportedModes: [], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'21': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'22': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'23': 
{ supportedModes: [ 0, 1, 3, 4 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'24': 
{ supportedModes: [ 0, 1, 3, 4 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'25': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'26': 
{ supportedModes: [ 0, 1, 3, 4 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'27': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'28': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 }, 
'29': 
{ supportedModes: [ 0, 1 ], 
mode: [Getter], 
value: [Getter/Setter], 
report: 1, 
analogChannel: 127 } } 
1508258664888 Board Closing.
```

and nodered is completely hung.

Waterbrain, 2017-10-17:
In case it could be of any use, this is the output of the contextbrowser in the global part.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

rwaldron, 2017-10-18:
This is super strange...

[image]

...Both `io` and `pins` properties are `null`!

nebrius, 2017-10-18:
I also noticed this at the end of the console output a few posts above:

```
1508258664888 Board Closing.
```

I suspect that is why nodered hung. Now, why did it do this though? I'm not sure. With the repl enabled, I would expect it to run forever, and I didn't see an exception which is the normal cause for exiting.

monteslu, 2017-10-18:
Hey everyone, sorry took so long to chime in.   node-red-contrib-gpio is my doing :)

Some background:  Node-red by default is kinda bad at running johnny-five.  Every time a flow is deployed in node-red a board instance is created.  In a single node vm, this is problematic because of leaks like event listeners or timers in components sticking around.
I made node-red-contrib-gpio with the intention of rectifying that by cleaning things up between node-red flow deployments and doing things like disabling the repl.  For the most part this works ok.  
Where things might run into problems with pin mappings is that to match the idioms of some existing node-red nod I just pass through whatever pins are specified in the UI directly to whatever IO class is selected and call the IO Classes's methods directly.  Inside of the j5 node, a script is evaluated that can use j5 components in the usual manner.

All that said, I'd start to debug this problem by removing node-red from the picture and just trying to use j5 and raspi-io directly together in node to see if things are cool.  If that works, the culprit is either node-red or node-red-contrib-gpio and we could open an issue over here:  https://github.com/monteslu/node-red-contrib-gpio

rwaldron, 2017-10-26:
> I'd start to debug this problem by removing node-red from the picture and just trying to use j5 

That's how I've been working on this, and I'm not able to reproduce the problem.

Waterbrain, 2017-10-26:
I'm out of my house for a trip. Also my mac has broken. 
I hope to be up and running by next Tuesday. 
I have also had problems with arduino with firmataplus. It also hangs node when you put a pin to read analog inputs and this pin does not support this type of input. I have not found how to name the analog arduino pins.
If you can think of any test that I can do or checklists, I will be happy to collaborate in whatever you deem appropriate.

kjetilei, 2017-11-14:
Im experiencing the same/similar issue with Node-red restarting randomly from time to time due to an uncaught error: TypeError: Cannot read property 'pinMode' of null.

@monteslu: I have opened [issue #17](https://github.com/monteslu/node-red-contrib-gpio/issues/28) to describe the case but only just now noticed this issue that seems to be related.

I have noticed that the uncaught exception only seems to happen after a while (a couple of minutes to many hours) if I leave the Node-red Dashboard open (where I'm monitoring a couple of the Arduino pins). But due to the random nature of when the reset is happening and the lack of log information, I haven't identified the root cause yet. And I'm still on an older version of Node-red that doesn't show the variable values (I haven't dared upgrading Node-red due to the complexity of my setup).

@Waterbrain: do you also use the Dashboard node?

Please let me know if I can be of any assistance and if anyone have any pointers on how to solve this.

Waterbrain, 2017-11-15:
Hello @kjetilei:
Effectively, I use the Dashboard node.
What seems strange to me is that using the nodes:
[image]
 y 
[image]
with these board definitions:
[image]

and
[image]
they work perfectly.

closed 2018-11-17 by stale[bot]