rwaldron/johnny-five #1397: Destroy component

rudacs, 2017-10-08:
I have a button (input) that I declare on a pin and in another moment it becomes a led (output) for example.

Except when creating the button there is no way to disable it or destroy it.
There is no disable or destroy feature to build another component on top of the pin.

Does anyone know any way?

rwaldron, 2017-10-09:
> I have a button (input) that I declare on a pin and in another moment it becomes a led (output) for example.

Can you explain how this happens with the physical hardware?

dtex, 2018-01-01:
Hi @rudacs just wanted to follow up on this. Can you tell us what is happening in the physical world? Is an end user expected to disconnect a button and wire up an LED on the same pin?

dtex, 2018-01-16:
Hi @rudacs, 

Since we haven't heard back I'm going to close this issue. Feel free to re-open any time.

closed 2018-01-16 by dtex