rwaldron/johnny-five #1402: Odroid XU3/4

chingling123, 2017-10-18:
Hi,

How can I use J5 with Odroid XU3 or XU4?

Thanks

dtex, 2017-10-18:
Hi @chingling123 

An [IO plugin](https://github.com/rwaldron/io-plugins) would need to be developed for the Odroid in order for it to work with Johnny-Five. Really that just means someone with the need for Odroid support decides they want to contribute to the J5 eco-system.

There are some node helpers for Odroid on NPM so it's a partially explored platform already.

chingling123, 2017-10-18:
@dtex Hi...

How can I create a IO plugin? Have some tutorial?

Thanks

dtex, 2017-10-19:
[This article](https://bocoup.com/blog/intel-galileo-javascript-nodejs) is the closest thing to a tutorial that I know of. 

In a nutshell, what you have to do is replicate the firmata.js surface API which is documented [here](https://github.com/firmata/firmata.js). Just replace all those properties and methods with code that works on the Odroid.

It's not trivial, but pull it off and you will be an expert at javascript on the Odroid.

chingling123, 2017-10-24:
@dtex Thanks... I'm read the article... but I need another help, I'm using linux-io to customize for odroid I'm getting some successful but I can't find documentation to enable stepper method from Johnny-five and PWM, do you have some link that I can read this?

closed 2017-10-24 by chingling123