rwaldron/johnny-five #1406: Board.detect should be throttled

mrmiller, 2017-10-25:
If there are no valid serial ports available, Board.detect will spin with repeated calls to itself and use 100% CPU. Detect should be called on a timer instead so that the calls are throttled to a reasonable rate.

dtex, 2018-01-01:
Sorry nobody has responded before now. 

This seems like a reasonable thing to do. There is also an attempt counter that is not being used (as far as I can tell). Would you like to submit a PR?

@reconbot Do you have any words of guidance on this?

[Relevant code block](https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L32-L102)
[Offending line](https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L91)

reconbot, 2018-01-01:
No events to hook into or anything. Polling every 200ms after the first scan is probably more than enough.

mrmiller, 2018-01-02:
Easiest solution would probably be to add a time.sleep call before calling Serial.Detect again. As @reconbot said, 200ms is more than frequent enough.

Using the attempt counter could have a similar effect but we would still want the sleep call, otherwise the client code would have to implement the throttling. I imagine Serial.detect would raise an exception if the optional max attempt count was exceeded. On the plus side, the client would have an indication that the device wasn't found rather than busy-waiting in the object constructor. It's currently pretty opaque, but this is a separate issue.

Should I put together a PR for the sleep call? Should be a one-liner.

dtex, 2018-01-02:
@rwaldron will have thoughts (and probably a better solution) so hold off for now but I think it can be non-blocking (setTimeout), give up after 10 attempts, and then call [board.fail()](http://johnny-five.io/api/board/#logging).

mrmiller, 2018-01-02:
Non-blocking seems like the best design decision overall. My only worry is that change would  break existing code that may be implicitly relying on the current blocking behavior.

reconbot, 2018-01-02:
The serial port is listed async, no way to be sync about it.

mrmiller, 2018-01-03:
You're absolutely right—serialport.list is at least structured as async (though I couldn't find a guarantee that any given platform implementation is async). In that case, it's already (likely) non-blocking and we should probably just throttle it with sleep.

As for finishing implementing the attempt counter, limiting the number of retries by default would certainly change existing behavior. As of now, the behavior is to just use an excessive amount of CPU while the device isn't connected but it will detect it if and when it shows up. With a default attempt limit, that would no longer be the case.

You all are much more familiar with the codebase and it's direction so I defer to your judgement on it. Let me know if you'd like me to implement something.

reconbot, 2018-01-03:
I can guarantee it's always async on all platforms since version 4 (and
probably before). 👍

dtex, 2018-04-29:
Closed with #1456

closed 2018-04-29 by dtex