rwaldron/johnny-five #1407: Stepper: Runstate / Position tracking

Triadager, 2017-10-26:
Hey guys!

Sorry to be a pain but googling and experimenting for hours didn't get me anywhere. 

I'm trying to control some steppers with an Arduino which gets its commands via Firmata. I want to implement some check whether to stepper is already moving if I issue another command. I figured this would be done with checking out the steppers "Runstate", but I couldn't get it to work at all. Did I misunderstand to what the "Runstate"-table on the stepper API page is referring to? If not, how can I access it? Could you maybe provide some example code on how to use these "Runstates"?

Another question regarding steppers: If I issue a different command while the stepper is moving, it'll interrupt the motion and the new command will be executed. Is there any possibility to track how many steps it already took? I'm using stepper driver boards like the A4988, so I'm thinking about just using another pin to monitor the pulses to the step pin, but I'm not sure if that's the most elegant solution. Any ideas?

Kind regards,

Triadager

dtex, 2017-10-26:
Hi @Triadager 

I will try and help. I believe those runstate values were inserted for future functionality that was never implemented. They don't appear to do anything. I could be just looking in the wrong place, but I don't think that's the case here. There is an isRunning state but it is private. You could create your own isRunning variable that is set to true whenever you call step, and set to false in a step callback.

> Is there any possibility to track how many steps it already took?

Not yet. There is a new version of [configurableFirmata](http://firmatabuilder.com/) that adds a new AccelStepperFirmata feature that when coupled with [this pull request](https://github.com/firmata/firmata.js/pull/179) for firmata.js will let you know actual position reached, even when the stepper is interrupted by another command. I expect that PR to land soon. It is just waiting to be blessed by @rwaldron.

The new version of the Johnny-Five stepper class that supports AccelStepperFirmata is in progress but likely won't be ready for weeks, but you can totally use the firmata.js stepper stuff directly on the board.io object. It's much higher level than the other firmata methods so it feels very J5'ish when using it.

dtex, 2017-10-29:
I'm going to close this. If you have more questions, feel free to reopen, or open a new issue on the applicable repo.

closed 2017-10-29 by dtex