rwaldron/johnny-five #1408: How can I read temperature from a MAX31855K Thermocouple?

jazzpeh, 2017-11-01:
I am trying to read temperature output from my connected thermocouple sensor. However, I wasn't able to get any readings as the thermocouple is actually connected to multiple pins. 

Here's the link to the setup that I did: https://learn.sparkfun.com/tutorials/max31855k-thermocouple-breakout-hookup-guide/all

Thanks!

fivdi, 2017-11-01:
The MAX31855K is an SPI (Serial Peripheral Interface) device but Johnny-Five currently doesn't support SPI.

Relates issues:
https://github.com/rwaldron/io-plugins/issues/3
https://github.com/rwaldron/johnny-five/labels/Protocol%3ASPI

> Are there any ways to make this work?

It may be possible to bit bang SPI using [Pin](http://johnny-five.io/api/pin/) objects.

dtex, 2018-01-01:
Two paths you could pursue to add support:

1. Contribute to firmata and add SPI https://github.com/firmata/arduino/issues/245
2. Create a component backpack for this device http://johnny-five.io/news/introducing-i2c-component-backpacks/

dtex, 2018-01-11:
Hi @jazzpeh ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for the MAX31855K there.

closed 2018-01-11 by dtex