rwaldron/johnny-five #1409: GPIO pins on Raspberry Pi Zero

zoobot, 2017-11-03:
Hi 
I want to use the moto zero motor controller with the raspberry pi zero.
Can I just use any GPIO pin or do I need to use PWM pins with raspi-io?
Thanks!
https://github.com/nebrius/raspi-io/wiki/Pin-Information
https://thepihut.com/products/motozero

rwaldron, 2017-11-20:
> I just noticed raspi-io says it is fermata compatible. I am confused about that, isn't the point of using raspi-io to use the raspberry pi GPIO pins directly instead of going through an fermata arduino or something?..

It's interface "compatible" with https://github.com/firmata/firmata.js

> With the python GPIO lib it looks like this: ...

Which Raspberry Pi are you using? 

> https://cdn.shopify.com/s/files/1/0176/3274/files/MotoZero_User_Guide_1.2.pdf

This document includes the following: 

> All examples in this document use the GPIO.BCM mode 

We'll have to ask @nebrius what impact this has on raspi-io

nebrius, 2017-11-20:
Related: https://github.com/nebrius/raspi-io/issues/100

> All examples in this document use the GPIO.BCM mode

Oh pin numbering on the Raspberry Pi, will you always be so complicated? /sarcasm

There are a variety of ways to refer to pin numbers on the Raspberry Pi, and each library takes a slightly different approach to how to number them.

Raspi IO follows the [Wiring Pi](https://projects.drogon.net/wiringpi-pin-numbering/) pin numbering scheme, but adds the ability to alias pins on top of this. The base number that Raspi IO uses is the _Wiring Pi_ number, which is neither the GPIO.BCM number _nor_ the GPIO.Board number. That said, you can use the BCM number in the format `GPIO##` or the Board number in the format `P#-#`.

In this case though, as we mentioned in the Raspi IO issue, there seems to be some sort of issue with aliasing though in this specific API. I haven't had a chance to test it yet though, so I'm not sure. I'll try and run some tests later today.

zoobot, 2017-11-20:
Thanks for answering! 

I am using the raspberry pi zero w.
Sort of got the pins working with the motozero. 
Didn’t get motor.forward to work, only reverse works. If I switch the physical motor wires, reverse works either way so I think something software is not quite right. 

//works with reverse for motozero/raspberry pi zero
const raspi = require('raspi-io');
const five = require('johnny-five');
const board = new five.Board({
  io: new raspi({ enableSoftPwm: true, includePins: ['GPIO24', 'GPIO27', 'GPIO5', 'GPIO6', 'GPIO22', 'GPIO17', 'GPIO23', 'GPIO16', 'GPIO12', 'GPIO13', 'GPIO18', 'GPIO25']})
});

board.on('ready', () => {
  //pins for MotoZero and Raspberry Pi Zero W
  var motor1 = new five.Motor(['GPIO24', 'GPIO27', 'GPIO5']);
  var motor2 = new five.Motor(['GPIO6', 'GPIO22', 'GPIO17']);
  var motor3 = new five.Motor(['GPIO23', 'GPIO16', 'GPIO12']);
  var motor4 = new five.Motor(['GPIO13', 'GPIO18', 'GPIO25']);

  //console.log("Go dog go");
  //motor1.forward(200);
  //motor1.start(255)
  // board.wait(5000, function () {
  //  motor1.stop();
  // });

  console.log("Back dog back");
  motor1.reverse(200);
  board.wait(5000, function() {
    motor1.stop();
  });
});

I ran out of time so ended up using python version with Flask instead of Node/Express/raspi-io/johnny-five. Would love to test with Johnny-five/express though because I am having a timeout issue with the python version, it would be helpful to rule out Flask as the problem. Not sure if timeout is Nginx load balancing issue, websockets disconnect issue, or something with Flask.

rwaldron, 2017-11-20:
> , there seems to be some sort of issue with aliasing though in this specific API.

`Motor` class?

rwaldron, 2017-11-20:
For this...

> Didn’t get motor.forward to work, only reverse works. 

Based on this...

```js
const raspi = require('raspi-io');
const five = require('johnny-five');
const board = new five.Board({
  io: new raspi({ enableSoftPwm: true, includePins: ['GPIO24', 'GPIO27', 'GPIO5', 'GPIO6', 'GPIO22', 'GPIO17', 'GPIO23', 'GPIO16', 'GPIO12', 'GPIO13', 'GPIO18', 'GPIO25']})
});

board.on('ready', () => {
  //pins for MotoZero and Raspberry Pi Zero W
  var motor1 = new five.Motor(['GPIO24', 'GPIO27', 'GPIO5']);
  var motor2 = new five.Motor(['GPIO6', 'GPIO22', 'GPIO17']);
  var motor3 = new five.Motor(['GPIO23', 'GPIO16', 'GPIO12']);
  var motor4 = new five.Motor(['GPIO13', 'GPIO18', 'GPIO25']);
```

- Perhaps you might not want all three pins to be PWM? (I have no idea, but I would check any documentation available)
- Did you try this test? http://johnny-five.io/api/motor/#situations-where-invertpwm-true-is-required

zoobot, 2017-11-20:
It looks like with enableSoftPwm, it just enables it as a blanket across any pins in includePins? I am not sure how to enable it only for certain pins. Some of those pi zero pins do have hardware PWM if I remember correctly...
https://cnpmjs.org/package/raspi-io <https://cnpmjs.org/package/raspi-io>

haven’t tried that test, I will do it this week. I would guess that invertpwm is not necessary if motor.reverse worked...
> Did you try this test? http://johnny-five.io/api/motor/#situations-where-invertpwm-true-is-required <http://johnny-five.io/api/motor/#situations-where-invertpwm-true-is-required>

I assumed the 3rd pin was enable after looking at the comments in motor.js inside the node lib but I am not really sure...
var motor1 = new five.Motor(['GPIO24', 'GPIO27', 'GPIO5']);

I couldn’t get this to work, I got errors about ‘pins’ undefined or something.
new five.Motor({
  pins: {
    cdir:'GPIO24',
    dir:'GPIO27',
    enable: 'GPIO5' 
  }
});

I did not try this, should I? 
new five.Motor({
  pins: {
    pwm:'GPIO24',
    pwm:'GPIO27',
    enable: 'GPIO5' 
  }
});

>

nebrius, 2017-11-20:
> Motor class?

Maybe? I can't say for sure till I put a debugger on it.

nebrius, 2017-11-20:
> includePins: ['GPIO24', 'GPIO27', 'GPIO5', 'GPIO6', 'GPIO22', 'GPIO17', 'GPIO23', 'GPIO16', 'GPIO12', 'GPIO13', 'GPIO18', 'GPIO25']

I just noticed this part in the constructor. Is there a reason you only wanted to include a subset of all pins? Normally this option isn't necessary except in very rare edge cases.

zoobot, 2017-11-20:
> includePins: ['GPIO24', 'GPIO27', 'GPIO5', 'GPIO6', 'GPIO22', 'GPIO17', 'GPIO23', 'GPIO16', 'GPIO12', 'GPIO13', 'GPIO18', 'GPIO25']
>
> I just noticed this part in the constructor. Is there a reason [...]

Just thought it was necessary to access those pins. The wording(except for ‘optional') makes it sound like the pins will be unavailable if I did not do that?
It was late and I was just mashing keys like a monkey.

includePins (optional)	Array<Number|String>	A list of pins to include in initialization. Any pins not listed here will not be initialized or available for use by Raspi IO
>

nebrius, 2017-11-20:
> Just thought it was necessary to access those pins. The wording(except for ‘optional') makes it sound like the pins will be unavailable if I did not do that?
It was late and I was just mashing keys like a monkey.

🤔  good point. I'll update the README, thanks!

nebrius, 2017-11-22:
I finally had a chance to stick a debugger on the [code specified here](https://github.com/nebrius/raspi-io/issues/100#issuecomment-342628966), and I figured out what's going on.

At [this line of code](https://github.com/rwaldron/johnny-five/blob/master/lib/motor.js#L600), `this.pins.pwm` is `undefined`, which makes sense given that `pwm` was never specified as an option to the Motor constructor. The code crashes because the Motor constructor called `this.io.pinMode(undefined, 3)`. This is the expected behavior for `pinMode`, but is this the expected way/best way for Johnny-Five to handle this situation?

rwaldron, 2017-11-27:
> At [this line of code](https://github.com/rwaldron/johnny-five/blob/master/lib/motor.js#L600), `this.pins.pwm` is `undefined`, which makes sense given that `pwm` was never specified as an option to the Motor constructor

Yep, that would do it. @dtex we should update the docs to include a table of required properties when `pins: { ... }` is used.

dtex, 2017-11-28:
Sure. How does this look:

* **pins object**     An object with pins required for the motor controller being used

 | Pin name | Required | Description |
 |-----------|----------|------------|
 | pwm         | Yes          | Controls motor speed |
 | dir             | Yes         | Controls motor direction |
 | cdir           | No           | Only necessary for 3-pin motor controllers. Is used in conjunction with dir to control direction and braking |
 | brake        | No           | Engages high impedance braking. Only a few motor controllers have a brake pin |

It would appear under [parameters](http://johnny-five.io/api/motor/#parameters)

nebrius, 2017-11-29:
What do you all think about adding a check in J5 as well so that if the required pins aren't specified, J5 throws a nice human-readable error, e.g. `Required "pwm" pin property not specified`, as opposed to letting Raspi IO throw with the generic (and not really helpful) error of `Unknown pin "undefined"`?

dtex, 2017-11-30:
Yes, I think that's a good idea. I'll try and get a PR in today.

dtex, 2018-01-01:
The PR has been merged and I've updated the wiki so I'm closing this issue

closed 2018-01-01 by dtex