rwaldron/johnny-five #1410: REQUEST: audio chip support Yamaha YM3812 and Commodore 6581

pappavis, 2017-11-05:
I'd like to modify the Piezzo Buzzer so as to use a MOS technology [6581 "SID"-chip](https://www.youtube.com/watch?v=qnngOMU8Wp0) or Yamaha [YM3812 OPL2](https://www.youtube.com/watch?v=ZwfsXiw3Ce8) sound support.

The **[piezzo.js](https://github.com/rwaldron/johnny-five/blob/master/lib/piezo.js)** already has the basic blocks, however one need to communicate with the sound chip.  For example, the sonar sensor communication.

I suppose a "l/b6581sid.js" would do the trick, however, how to get the arduino .cpp-library translated to johnny-five?  Any ideas?

Regards,
Michiel Erasmus / Easylab4kids

rwaldron, 2017-11-20:
@pappavis The code in piezo.js is not capable of producing actual tones, due to timing sensitive constraints and limitations of JavaScript itself. For new, more specific functionality I recommend first creating a plugin to prove out the idea; for example: https://github.com/rwaldron/j5-button-gestures or https://github.com/rwaldron/j5-ds-touch

pappavis, 2017-11-20:
I'd hope to get SIDuino working in Johnny-five.
https://playground.arduino.cc/Main/SIDuinoI2c

pappavis, 2017-11-22:
@rwaldron 
I am clueless reading the ds_touch_slave digitalinput/outputs in javascript .

What i wanna do is mimic Pizzo functionality such as Play() so can interchange piezo for OPL2 without code rewrite.

```
   var objPiezo_1;  //piezo object

   objPiezo_1_playing = true;
   objPiezo_1.play({
     song: "E - D - C - D - - E E E D D D - E - G ",
     beats: 4 / 4,
     tempo: 250});
  objPiezo_1_playing = false
```

I wanna play for example this --> https://github.com/DhrBaksteen/ArduinoOPL2/blob/master/examples/SimpleTone/SimpleTone.ino
[which isnt as straihtforward as your Pizzo play function]
In above example 2x libraries are included.

Furthermore. 
* Wheres the link from _readY()_  in ds_touch_slave.ino to where it can be used in javascript plugin code? is it sent to a I2C adress?
* What must be done to translate C++ opl2.setTremolo() --> plugin, so it becomes javascript?

How I think tot do it;
 - Use your PingFirmata.ino and add #include OPL2.h and SPI.h to it.
 - Make plugin. Sumhow get javascript setTremolo so it can be received by Firmata.  **HOW?**
 - Let it make a simple bell-sound.

dtex, 2018-01-01:
@pappavis 

> Sumhow get javascript setTremolo so it can be received by Firmata. HOW?

For the OPL2 you will need SPI which has been [proposed but is not yet supported by firmata](https://github.com/firmata/protocol/blob/master/proposals/spi-proposal.md). Contributing to that proposal would be the first step for that device.

You've referenced two different approaches to SID. There's the [sid-arduino-lib](https://github.com/cybernesto/sid-arduino-lib) which is an emulator that uses nothing but an Arduino and the [SIDuino i2c SID-emulator shield](https://playground.arduino.cc/Main/SID-emulator) which is an I2C device that works with an Arduino. 

If you want to use the sid-arduino-lib you might consider creating an [I2C Component Backpack](http://johnny-five.io/news/introducing-i2c-component-backpacks/) for Johnny-Five.

The SIDuino shield *might* be the easiest JS solution to implement as you can completely craft the i2c messages in node and push them over via the existing i2c write methods in firmata. The only thing I'm worried about is how many i2c messages you might need to send in short order. They can get jammed up. The code you would need to port is in siduino_midi_i2c.pde which can be found in a [zip file on this page](https://playground.arduino.cc/Main/SIDuinoI2c).

As for how to make it fit in the Johnny-Five class schema, a plug-in is the way to test the idea out and I don't think it's something that could ever fit under piezo as these devices can do so much more.

dtex, 2018-01-11:
Hi @pappavis 

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just don't know when someone will have the time to add this (admittedly very cool) new functionality. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added a new class request for "Sound Generator/Synthesizer" there.

closed 2018-01-11 by dtex

pappavis, 2018-01-21:
> Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just don't know when someone will have the time to add this (admittedly very cool) new functionality. Rather than leave it languishing as an open issue we have created a Requested Features <x-msg://11/wiki/Requested-Features> page and added a new class request for "Sound Generator/Synthesizer" there.

Thanks!

Михель Эрасмус
Tel: +31638369260