rwaldron/johnny-five #1411: LCD not initialising properly

Overwatchh, 2017-11-09:
Hi!

When I connect the RPI with a LCD display to power and run a NodeJS program that uses J5 to communicate with the LCD this happens:

[image]

In the top lone the text is as bright as it should be, but all the other pixels are light half way, while the bottom line is completely blank.

Before I used a Python script to control the LCD and it worked fine. Even now, starting that python script just for a few seconds fixes the issue and the display works fine after that (even with NodeJS and j5).

Here's all the code that isn't writing or moving the cursor:
`
lcd = new five.LCD({
	pins: ["P1-7", "P1-11", "P1-12", "P1-13", "P1-16", "P1-15"]
});
lcd.off();
lcd.on();
`

Another thing: If I remove the lcd.off() and lcd.on() it all works as before, but weird characters are displayed on the screen.

I'm using:
Raspberry Pi 3 Model B Rev 1.2
NodeJS 8.9.1
j5 0.11.6
raspi-io 8.1.0

Thanks in advance for any help :)

fivdi, 2017-11-14:
The program posted above doesn't actually print anything to the LCD. Can you post a complete program that can be used to reproduce the problem please?

Please also post a link to a description of LCD being used.

> In the top lone the text is as bright as it should be, but all the other pixels are light half way, while the bottom line is completely blank.

This may be related to how the contrast pin on the LCD is being controlled. How is the contrast pin being controlled? Is there a contrast pot like to one shown at http://johnny-five.io/api/lcd/#component-initialization. If the pot is twisted does anything improve?

Overwatchh, 2017-11-16:
Sorry for the late response. I've been very busy these past few days.

I now have new (but identical) hardware, and the rows are light up the same as before but no characters are showing. Same as before, initializing the display with the Python script fixes the issue.
I don't have the exact LCD (I didn't assemble the hardware myself), but it is HD44780 compliant.

The contrast pin is wired correctly and I tried turning it, but it just changes the contrast of the top row. Characters are always invisible and the bottom row is always completely blank.

I wrote this just for debugging. The other program is too large to post, but it works the same way:

`
"use strict";
const Raspi = require("raspi-io"),
    five = require("johnny-five");

const board = new five.Board({
    io: new Raspi(),
    repl: false,
    debug: true
});
let lcd;
function writeToLCD (row1, row2) {
    lcd.clear();
    lcd.cursor(0, 0).print(row1);
    lcd.cursor(1, 0).print(row2);
}

let number = 0;
board.on("ready", function() {
    lcd = new five.LCD({
        pins: ["P1-7", "P1-11", "P1-12", "P1-13", "P1-16", "P1-15"]
    });
    lcd.off();
    lcd.on();
    setInterval(function () {
        writeToLCD(''+number, '' + ++number)
    }, 1000)
});

`

fivdi, 2017-11-16:
I just wired up a 16x2 LCD and ran the above program exactly as it is. The problem can be reproduced every second or third run of the program. If I end up having some free time at the weekend I'll take a closer look and try to figure out what's going on.

fivdi, 2017-11-18:
In `node_modules/johnny-five/lib` there is a file called `lcd.js` which contains two calls to function `sleepus`. The first call which is supposed to wait for 1 microsecond is [here](https://github.com/rwaldron/johnny-five/blob/v0.11.6/lib/lcd.js#L841) and the second call which is supposed to wait for 37 microseconds is [here](https://github.com/rwaldron/johnny-five/blob/v0.11.6/lib/lcd.js#L850). Please edit `lcd.js` on your Raspberry Pi and replace both of these calls to `sleepus` with a call to `sleep(1)` which should wait for 1 millisecond:
```js
sleep(1);
```
Then give it a try to see if it works.

Please also try out the below program. It's a variant of your program form above that initializes the LCD, waits a second, displays text on the LCD, waits a second and repeats the process. Let it run for about 100 iterations while observing the LCD to ensure that everything works correctly.
```js
"use strict";
const Raspi = require("raspi-io"),
    five = require("johnny-five");

const board = new five.Board({
    io: new Raspi(),
    repl: false,
    debug: true
});
let lcd;
function writeToLCD (row1, row2) {
    lcd.clear();
    lcd.cursor(0, 0).print(row1);
    lcd.cursor(1, 0).print(row2);
}

let number = 0;
board.on("ready", function() {
    lcd = new five.LCD({
        pins: ["P1-7", "P1-11", "P1-12", "P1-13", "P1-16", "P1-15"]
    });
    lcd.off();
    lcd.on();
    setInterval(function () {
        writeToLCD(''+number, '' + ++number)
    }, 1000)
});
```

Overwatchh, 2017-11-18:
Progress!
After the first change it started working the way it did with the earlier hardware. The bottom row is  empty and in the top row the text is visible and was fully dark as it should be, but the rest of the pixels are half way dark. The text is visible, but not very well. Especially if I look at it straight on.

Turning the contrast dial up makes the background pixels darker making the text more difficult to read. Turning it down makes the text more clear up to the point when the text becomes invisible (i guess too low contrast), but until that point the text is always the same.

I tried your program, but the result is the same as described above.

fivdi, 2017-11-18:
That sounds like nothing much has changed to me, is this correct?

Overwatchh, 2017-11-18:
Yes.
The progress is that on the new hardware the text is visible as it was with the old hardware. It is strange since they should be identical.

fivdi, 2017-11-18:
Try increasing the values in the calls to sleep [here](https://github.com/rwaldron/johnny-five/blob/v0.11.6/lib/lcd.js#L681-L685), [here](https://github.com/rwaldron/johnny-five/blob/v0.11.6/lib/lcd.js#L917) and [here](https://github.com/rwaldron/johnny-five/blob/v0.11.6/lib/lcd.js#L923) to see if it improves things.

Add additional calls to sleep directly before and after [this](https://github.com/rwaldron/johnny-five/blob/v0.11.6/lib/lcd.js#L681-L685) block of code. Start with `sleep(16)` before the block and `sleep(4)` after the block.

fivdi, 2017-11-18:
So everything works correctly on the new hardware?

Overwatchh, 2017-11-18:
At first with the new hardware it was like the picture, but without the text. With the changes I made earlier it is now just like the picture.

I quadrupled the values and added the extra sleep calls, but it's still the same.

fivdi, 2017-11-18:
If the test programs from above are being used I'm afraid I'm out of ideas.

Overwatchh, 2017-11-18:
:(

Maybe this will help. A previous version of the program used this LCD driver for python. Now, if I run this so that it initializes, stop it, and then run the program (mine or yours, even without the extra sleep) everything works the way it should. There must be something in this driver that changes some setting or something and fixes the issue.
[lcddriver.zip](https://github.com/rwaldron/johnny-five/files/1484522/lcddriver.zip)

fivdi, 2017-11-18:
One difference is that the python driver has a 200 millisecond delay after the clear in __ init __. A sleep(200) could be added after the [clear in lcd.js](https://github.com/rwaldron/johnny-five/blob/v0.11.6/lib/lcd.js#L702) to see if it helps.

Overwatchh, 2017-11-18:
It doesn't help :(

Overwatchh, 2017-11-18:
Ok so I tried [this library](https://www.npmjs.com/package/lcd) and it seems to work perfectly. I just copied the example and set the correct pins, columns and rows. 

And just after that I found out it was yours :D

So I guess I should use this instead of j5 to control the LCD. The other things I use j5 for are just some relays and a LED (which I have to control by setting pins manually because of hardware limitations)
At this point I'm thinking of just switching away from j5. What do you think?

fivdi, 2017-11-19:
I guess switching over to the lcd module is an option. On the other hand I wish I could figure out why J5 LCD doesn't function correctly on your system. The same J5 code works fine on my Raspberry Pi.

Overwatchh, 2017-11-23:
I'll switch the lcd module, but if you want to pursue this I can help

closed 2019-01-03 by stale[bot]