rwaldron/johnny-five #1413: Installation instructions inadequate

peetj, 2017-11-15:
Hi

Firstly many thanks for the work you have done on Johnny-Five
I was 'getting my Arduino board ready' to your instructions and found that they were slightly inadequate.

For example you say install the arduino package. Do you mean nodejs? Do you mean a global install?
Then it says run arduino --install-library "Firmata,Servo"
All I get is 'Command not Found'

You need to add some context and make it clearer otherwise you are going to confuse a whole lot of people.

I am no newbie but your instructions have left me confused.

Many thanks

Pete...

pappavis, 2017-11-15:
What I did;
1. Download and install [nodeJS](http://nodejs.org)
2. Open a commandline (Windows), or shell (Mac, Linux, Raspi).
3. On the commandline do;
`npm install johnny-five`
4. Start the Arduino IDE, the open Examples --> Firmata --> StandardFirmata
5. Upload to the Arduino.
6. Create your j5 program such as helloworld.js
7. run using `node ./helloworld.js`

Now it must work.

rwaldron, 2017-11-20:
I'm not sure what instructions you're referring to? The ones in the readme contain everything you recommend in your second comment: 

- https://github.com/rwaldron/johnny-five#setup-and-assemble-arduino
- https://github.com/rwaldron/johnny-five#hey-you-heres-johnny
- https://github.com/rwaldron/johnny-five#example-programs

dtex, 2017-11-20:
I think the instructions he is talking about are on [this page](https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting). They do seem dated.

developer239, 2018-07-14:
@peetj +1 This might be a stupid question but how do I run Arduino from a command line? 

```
$ arduino: command not found
```

developer239, 2018-07-14:
@pappavis Thank you for your information. I finally see blinking LED 🙂

mven, 2018-08-25:
@pappavis Thank you! This worked for me, as I was having trouble with the command as well.

@rwaldron The Getting Started doc in the Troubleshooting section is where I can see some confusion. It's not clear if we should install the `arduino` package via NPM or from somewhere else. In the context instructions, it would make sense to install it from there, but after doing so the command `arduino` is not available resulting in the `command not found` error. Doing some more investigation, it looks like that package hasn't been updated in quite some time (last release was [2011](https://www.npmjs.com/package/arduino)). The getting started doc is a great starting point, but just needs a little bit of updating, but unless there's an `arduino` package that I'm not aware of that works as a CLI tool, then I'd suggest omitting that from the instructions for the time being and just use the IDE.

I can make that change to reflect @pappavis's more detailed instructions as I just tested it out and it's working nicely.

closed 2019-08-27 by stale[bot]