rwaldron/johnny-five #1414: TypeError: Cannot read property 'Board' of undefined

TheLLspectre, 2017-11-16:
Hi everyone !

I just want to know if someone can help me with this error, I don't understand what happened.
I try to work with TypeScript. Just with basic example, so i've this:

> import { five } from 'johnny-five';
import { Raspi } from 'raspi-io';
const board = new five.Board({
  io: new Raspi()
});

And i have this error: 

> Runtime Error
Cannot read property 'Board' of undefined
Stack
TypeError: Cannot read property 'Board' of undefined
    at Object.187 (http://localhost:8100/build/main.js:62:67)
    at __webpack_require__ (http://localhost:8100/build/vendor.js:55:30)
    at Object.249 (http://localhost:8100/build/main.js:179:75)
    at __webpack_require__ (http://localhost:8100/build/vendor.js:55:30)
    at Object.208 (http://localhost:8100/build/main.js:129:73)
    at __webpack_require__ (http://localhost:8100/build/vendor.js:55:30)
    at Object.188 (http://localhost:8100/build/main.js:113:70)
    at __webpack_require__ (http://localhost:8100/build/vendor.js:55:30)
    at webpackJsonpCallback (http://localhost:8100/build/vendor.js:26:23)
    at http://localhost:8100/build/main.js:1:1

What do you think ???

monteslu, 2017-11-16:
Looks like your import is wrong.
try `import five from 'johnny-five';`

TheLLspectre, 2017-11-16:
ok, i try this, it's really interesting but how it's possible, others imports are wrote like this, and it's ok why this one doesn't work !

So i think it goes inside but i've an another error:

> ypeError: exists is not a function
    at Function.getRoot (http://localhost:8100/build/vendor.js:152508:9)
    at bindings (http://localhost:8100/build/vendor.js:152410:32)
    at Object.<anonymous> (http://localhost:8100/build/vendor.js:151857:41)
    at __webpack_require__ (http://localhost:8100/build/vendor.js:55:30)
    at Object.<anonymous> (http://localhost:8100/build/vendor.js:151609:22)
    at Object.<anonymous> (http://localhost:8100/build/vendor.js:151612:30)
    at __webpack_require__ (http://localhost:8100/build/vendor.js:55:30)
    at Object.<anonymous> (http://localhost:8100/build/vendor.js:150526:17)
    at __webpack_require__ (http://localhost:8100/build/vendor.js:55:30)
    at Board.detect (http://localhost:8100/build/vendor.js:116964:20)

and i just write:

`const board = new five.Board();`
with the new import ...;

monteslu, 2017-11-16:
your code was using the new destructuring operator from ES6.  It allows you to pull properties out of  a module.  In this case you're referencing the `five` variable directly in your code, so you don't need to destructure. 
I'm not sure, but you may also need to do the same with your raspi-io import.

TheLLspectre, 2017-11-16:
Yeah!
with this :
`import five from 'johnny-five';
import Raspi from 'raspi-io';`
i can put :
`const board = new five.Board({
  io: new Raspi()
});
`
but also ....

> TypeError: fs_1.readFileSync is not a function
    at Object.<anonymous> (http://localhost:8100/build/vendor.js:149737:21)
    at Object.<anonymous> (http://localhost:8100/build/vendor.js:149826:30)
    at __webpack_require__ (http://localhost:8100/build/vendor.js:55:30)
    at new RaspiIO (http://localhost:8100/build/vendor.js:146538:15)
    at Object.187 (http://localhost:8100/build/main.js:66:9)
    at __webpack_require__ (http://localhost:8100/build/vendor.js:55:30)
    at Object.249 (http://localhost:8100/build/main.js:181:75)
    at __webpack_require__ (http://localhost:8100/build/vendor.js:55:30)
    at Object.208 (http://localhost:8100/build/main.js:131:73)
    at __webpack_require__ (http://localhost:8100/build/vendor.js:55:30)

I saw this one few times, i try to check it but if you know this one .... ;)

monteslu, 2017-11-16:
that one could be something going on in raspi-io.
Did you follow the instructions where it said to install it as a regular user and run it as root?

TheLLspectre, 2017-11-16:
hum, to install as a regular user, yeah i think so, but not sur for run it as root !

TheLLspectre, 2017-11-16:
i have no idea to how to fix it ...

rwaldron, 2017-11-20:
Thanks for the report and thanks @monteslu for helping to find a solution! Since the reported issue itself has been addressed (and isn't an issue with Johnny-Five), I'm going to close this.

closed 2017-11-20 by rwaldron