rwaldron/johnny-five #1416: DRV8833 Motor Driver

nofear87, 2017-11-19:
Is there support for the DRV8833 dual motor driver?

I try to get it working with the dual h bridge example. But no chance :(

Next Problem is, that I need more PWM then a raspberry pi 2b can offer, right?

Thank you!

ghtomcat, 2017-11-19:
I'm successfully using the Adafruit DRV8833 Breakout Board with an ESP8266. I'm using direct setPWM and setPin calls however, since the ESP8266 uses 10-bit PWM

rwaldron, 2017-11-20:
@dtex @nebrius

dtex, 2017-11-20:
I haven't user a DRV8833, but I would think the h bridge example would be the right one to follow. Can you tell me what is happening? 

Possible problems:
 * Using ```invertPWM: true``` when you don't need it. Some boards handle the logic inversion for you and don't need this property. The DRV8833 is a pretty humble device so I doubt this is the case.
 * No common ground. You must join the ground between the Arduino and the motor driver so that the devices have a common baseline for low (this is a common mistake).
 * Some other wiring problem.

If none of these are the problem, can you share your code and a picture of your wiring?

Deferring to nebrius re: pi questions.

rwaldron, 2017-11-20:
w/r to @dtex's wiring request, here's Pololu's reference: 

[image]

Fromt this page: https://www.pololu.com/product/2130 , under the section "Pinout":

PIN | Default State | Description
--- | ------------- | ------------
AIN1 | LOW | A logic input control for motor channel A. PWM can be applied to this pin.
AIN2 | LOW | A logic input control for motor channel A. PWM can be applied to this pin.
BIN1 | LOW | A logic input control for motor channel B. PWM can be applied to this pin.
BIN2 | LOW | A logic input control for motor channel B. PWM can be applied to this pin.

nebrius, 2017-11-20:
> Next Problem is, that I need more PWM then a raspberry pi 2b can offer, right?

If you pass in `enableSoftPwm: true` to the [Raspi IO constructor](https://github.com/nebrius/raspi-io#new-raspioptions), you can use any pin on the Raspberry Pi as PWM.

nofear87, 2017-11-21:
Thank you guys for all your usefull hints! I got it working. The problem was a to low voltage, so that the motors did not start running.

dtex, 2017-11-21:
Excellent! Have fun.

closed 2017-11-21 by dtex