rwaldron/johnny-five #1417: io.pinMode is not set when pwmRange is passed in servo.opts

dtex, 2017-11-25:
Blocking https://github.com/rwaldron/tessel-io/pull/21

fivdi, 2017-11-25:
@dtex do you mean in [this](https://github.com/rwaldron/johnny-five/blob/v0.11.7/lib/servo.js#L55-L59) code when either `servoConfig` or `pinMode` is called? If so, [according to the io-plugins specification](https://github.com/rwaldron/io-plugins#servoconfigoptions) `servoConfig(options)` may be called as an alternative to calling `pinMode`. In reality, I don't think `servoConfig(options)` is ever actually called but `servoConfig(pin, min, max)` is and by the looks of things as an alternative to calling `pinMode`. At least that's what I noticed when implementing [servoConfig(pin, min, max) in linux-io](https://github.com/fivdi/linux-io/blob/v0.8.0/lib/linux-io.js#L216-L227). There were scenarios where `servoConfig(pin, min, max)` was being called for pins that were not in `SERVO` mode so it was necessary to call `pinMode` inside `servoConfig(pin, min, max)`.

dtex, 2017-11-25:
@fivdi that’s exactly it. Thanks for explaining. The problem is with my servoConfig implementation in another io plugin. Thanks!

closed 2017-11-26 by dtex