rwaldron/johnny-five #1418: LCD clear can cause gibberish

wagenet, 2017-11-29:
I'm using the J5IK. I have the following app:

```js
var moment = require("moment");
var five = require("johnny-five");
var Tessel = require("tessel-io");
var board = new five.Board({
  io: new Tessel()
});

board.on("ready", function() {
  var monitor = new five.Multi({
    controller: "BME280"
  });

  var lcd = new five.LCD({
    //      RS    EN    D4    D5    D6    D7
    pins: ["a2", "a3", "a4", "a5", "a6", "a7"],
  });

  var lastUpdate = 0;

  monitor.on("change", function() {
    var now = Date.now();
    if (now > lastUpdate + 1000) {
      lcd.clear();
      lcd.cursor(0, 0).print(this.temperature.fahrenheit+"F");
      lcd.cursor(1, 0).print(this.hygrometer.relativeHumidity+"%");
      lastUpdate = now;
    }
  });
});
```

At first things work fine, but after a few updates, it starts printing gibberish to the LCD. If I remove the `lcd.clear()` it works correctly.

fivdi, 2017-11-29:
Does it work if this:
```js
  monitor.on("change", function() {
    var now = Date.now();
    if (now > lastUpdate + 1000) {
      lcd.clear();
      lcd.cursor(0, 0).print(this.temperature.fahrenheit+"F");
      lcd.cursor(1, 0).print(this.hygrometer.relativeHumidity+"%");
      lastUpdate = now;
    }
  });
```
is changes to this:
```js
  monitor.on("change", function() {
    var now = Date.now();
    if (now > lastUpdate + 1000) {
      lcd.clear();
      lastUpdate = now;
      setTimeout(function() {
        lcd.cursor(0, 0).print(this.temperature.fahrenheit+"F");
        lcd.cursor(1, 0).print(this.hygrometer.relativeHumidity+"%");
      }, 10);
    }
  });
```
If that doesn't help does increasing the timeout from 10 to 100 milliseconds help?

fivdi, 2017-11-29:
Here's what I think is happening here, however, I don't have a Tessel 2 so I'm doing quite a bit of guessing.

When a clear command is sent to an LCD it takes the LCD 1.52 milliseconds to execute that command. In order to give the LCD enough time to execute the clear command Johnny-Five has a 2 millisecond delay [here](https://github.com/rwaldron/johnny-five/blob/v0.11.7/lib/lcd.js#L917). During this time no other commands should be sent to the LCD.

This 2 millisecond delay is in the Node.js process which runs on the [Tessel 2](https://tessel.io/) Mediatek MT7620n application processor. However, the LCD isn't actually connected to the Mediatek MT7620n, it's connected to the Atmel SAMD21 Cortex-M0 coprocessor. To function correctly the 2 millisecond should really be on the Cortex-M0 coprocessor, not on the Mediatek MT7620n application processor.

I'm not familiar with the internals of tessel-io but if it uses asynchronous write-and-forget to communicate with the coprocessor  via SPI I can imagine scenarios where the 2 millisecond delay discussed above has completed before the LCD has actually started to execute the clear command. If this happens, tessel-io may start sending commands new commands to the LCD before the clear command has completed.

wagenet, 2017-12-01:
@fivdi the `setTimeout` avoids the issue. Your analysis seems reasonable to me, but I'm far from an expert on this stuff.

dtex, 2018-01-01:
It sounds like the advice from @fivdi worked so I'm going to close this.

If anyone wants to dig deeper and actually fix the issue, I think the thing to do is recreate the problem using just the [T2 Hardware API](https://tessel.gitbooks.io/t2-docs/content/API/Hardware_API.html) and solve from there.

closed 2018-01-01 by dtex

fivdi, 2018-01-02:
@dtex If the above analysis as to why the issue occurred is correct then it's more of a design flaw in how Johnny-Five handles timeouts than a T2 Hardware API issue. But as you mention it would be necessary to dig deeper to figure out what's actually happening.