rwaldron/johnny-five #1421: tinker board support

moeinrahimi, 2017-12-05:
does johhny-five supporting asus tinker board ?

fivdi, 2017-12-05:
To the best of my knowledge there isn't an [IO Plugin](https://github.com/rwaldron/io-plugins) for the Asus Tinker board. If this is something that you would be interested in implementing [Linux-IO](https://github.com/fivdi/linux-io) could be considered as a starting point.

dtex, 2018-01-11:
Hi @moeinrahimi 

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for tinker board support there.

closed 2018-01-11 by dtex

knaut, 2018-01-29:
Hi all,

Big fan of Johnny-Five and its wide variety of boards support. We have a project running on a Raspi 3 and upgraded to Tinker Board lately for its beefier performance. Since J5 doesn't support Tinker out of the box, I tried to give it a go myself.

Since the Tinker Board and Rpi 3's GPIO pinouts are supposed to be identical, I wrote a bare-bone program using the raspi-io plugin as-is (retyped, not copy-pasted, so apologies for any typos):

`var five = require('johnny-five');
var Raspi = require('raspi-io');

var board = new five.Board({
  io: new Raspi()
});

board.on('ready', function() {
  console.log('board ready!');
});`

Here's how far it got (terminal output, again retyped):

`Unknown board revision 0000, assuming Raspberry Pi 3 pinout.
1517189876366 Available Raspberry Pi-IO
2018-01-29 01:37:56 initPeripherals: mmap gpio failed (Operation not permitted)
/root/rootkri/node_modules/pigpio/pigpio.js:11
  pigpio.gpioInitialise();

Error: pigpio error -1 in gpioInitialise
  at initializePigpio (/root/rootkri/node_modules/pigpio/pigpio.js:11:12)
  at new Gpio (/root/rootkri/node_modules/pigpio/pigpio.js:25:3)
  at new DigitalOutput (/root/rootkri/node_modules/raspi-gpio/dist/index.js:79:25)
  at RaspiIOCore.value (/root/rootkri/node_modules/raspi-io-core/dist/index.js:553:38)
  at RaspiIOCore.pinMode(/root/rootkri/node_modules/raspi-io-core/dist/index.js:505:21)
  at /root/rootkri/node_modules/raspi-io-core/dist/index.js:427:17
  at Array.forEach (<anonymous>)
  at /root/rootkri/node_modules/raspi-io-core/dist/index.js:365:32
  at process._tickCallback (internal/process/next_tick.js:150:11)
  at Function.Module.runMain (module.js:703:11)`

After running the program a couple times as a normal user (`$ nodejs index.js` slightly different error about "modprobe", then with `sudo`), and getting the above error, I tried installing the program at `/root/` and running it as root (hence the "/root/rootkri" paths in the error message). Maybe dangerous, but no dice either way.

I've found this same error when running similar programs on a Raspberry Pi while the user doesn't have permissions for the GPIO. Usually, using "sudo" fixes that. Not so in this case for the Tinker Board.

Is getting the Raspi-IO plugin to work on Tinker Board simply a permissions issue for GPIO, despite the identical pinout, or is there something deeper going on?

We'd be interested in extending the Linux-IO plugin to leverage J5's API on the Tinker Board. For someone who knows their way around Javascript but not necessary Linux and embedded computing systems, any pointers on writing a plugin like this for Tinker Board?

fivdi, 2018-01-30:
@knaut 

> Is getting the Raspi-IO plugin to work on Tinker Board simply a permissions issue for GPIO, despite the identical pinout, or is there something deeper going on?

Under the covers `raspi-io` uses the [pigpio Node.js module](https://github.com/fivdi/pigpio) to access hardware on the Raspberry Pi. The `pigpio` Node.js module then uses the [pigpio C library](https://github.com/joan2937/pigpio) to access hardware. The `pigpio` C library was implemented specifically for BCM2835/6/7 processors, i.e., the processors used on the various variants of the Raspberry Pi. There is absolutely no chance of the `pigpio` C library functioning on an ASUS Tinker Board which has a Rockchip RK3288 processor.

> I'd be interested in extending the Linux-IO plugin to leverage J5's API on the Tinker Board. For someone who knows their way around Javascript but is relatively new to Linux and embedded computing systems, any pointers on writing a plugin like this for Tinker Board?

I'd suggest taking a look at [tiny-raspberry-pi-io.js](https://github.com/fivdi/linux-io/blob/master/example/raspberry-pi/tiny-raspberry-pi-io.js) which is a tiny IO Plugin for the Raspberry Pi. There are 7 example programs in the [directory](https://github.com/fivdi/linux-io/tree/master/example/raspberry-pi) together with `tiny-raspberry-pi-io.js`. Try to get the exampels to work on your Raspberry Pi. Study and try to understand the code in `tiny-raspberry-pi-io.js`. To get the I2C example to function you'll need to activate I2C on the Pi with the `raspi-config` tool.

After you understand `tiny-raspberry-pi-io.js` try to implement a similar tiny IO Plugin on the ASUS Tinker Board.

[BeagleBone-IO](https://github.com/julianduque/beaglebone-io) is a fairly complete IO Plugin. You could also study its code. It adds features like PWM, SERVO and ANALOG which Linux-IO doesn't support.