rwaldron/johnny-five #1422: Expander MCP23017, issue with read events when using the two ports simultaneously

dygoon, 2017-12-08:
hello,

HW: rasp PI 3
latest J5 version.

if you are registering read events for two pins with the same number on each port, say GPIOA-1 (pin 1) and GPIOB-1 (pin 9), the two read event will be generated with the same "digital-read-1" name. Therefore, whenever any of those two pins is read, the two registered callbacks will be called, resulting in an unexpected behaviour.

to fix this, the pin number passed as a parameter should be saved and used to generate the event

value: function(pin, callback) {
var orgPin=pin;
...
this.on("digital-read-" + orgPin, callback);
...
this.emit("digital-read-" + orgPin, value);

I hope it helps, I tested the fix and it works well.

best,
alex

original code----

digitalRead: {
      value: function(pin, callback) {
        var pinIndex = pin;
        var gpioaddr = 0;

        if (pin < 8) {
          gpioaddr = this.REGISTER.GPIOA;
        } else {
          gpioaddr = this.REGISTER.GPIOB;
          pin -= 8; // <----- PIN NUMBER IS MODIFIED HERE, GPIO A&B end up with the same event name
        }

        this.pins[pinIndex].report = 1;

        this.on("digital-read-" + pin callback);

        this.io.i2cRead(this.address, gpioaddr, 1, function(data) {
          var byte = data[0];
          var value = byte >> pin & 0x01;

          this.pins[pinIndex].value = value;

          this.emit("digital-read-" + pin, value);
        }.bind(this));
      }
    },
  },

rwaldron, 2017-12-30:
I'm so sorry it's taken me this long to respond. Your fix looks good and makes sense. I will be back to work next week and will set up some test hardware to confirm.

I appreciate your patience. I'm a new Dad this year and I've been struggling to stay on top of this project.

closed 2018-02-20 by rwaldron

rwaldron, 2018-02-20:
@dygoon fix landed and released! Again, sorry for the ridiculous delay :(

dygoon, 2018-03-03:
hello, no worries ! and congrats for being a dad! it's a different type of fun!
all the best!