rwaldron/johnny-five #1429: Offset considered twice

jbdesbas, 2017-12-30:
Hello,

On the Servo class, when a time is specified on the Servo.to() method, the offset is considered twice
eg with -10 offset : 

> 
> servo.to(80)
> servo.value //70
> 
> servo.to(80,100)
> servo.value //60

Observed with ESP8266/EtherPortClient and PCA9685 as servo controller

dtex, 2017-12-30:
Thanks @jbdesbas 

The code looked right at first blush so I'm writing some tests to replicate. I have this nagging feeling that this used to be an issue and was fixed. Can you tell me what version of J5 you are using?

dtex, 2017-12-30:
@jbdesbas I've replicated and fixed the bug. I'll get a PR up there shortly.

dtex, 2017-12-30:
@jbdesbas I expect this will land in the next day or so. In the meantime can I ask how you are using offset? I just like to hear about the various use cases so I can be considerate of them in future code changes.

jbdesbas, 2017-12-30:
Hi @dtex , thank you for everything !
J5 version is 3.5.2
I use offset for a robot arm, because this is a difference between the real angle and the "theorical" angle I used in trigonometrical calculations (for moving servos in order to reach polar or cartesian coordinates.)
I just started the project, so maybe is not the best approach.

dtex, 2017-12-30:
Oh cool. I think you are on the right track. I have a 3-DOF [MeArm](https://www.thingiverse.com/thing:360108) that I've been meaning to wire up with Johnny-Five. 

My plan is to extend a library called [Tharp](https://github.com/dtex/tharp) that I made for my hexapod. The library and documentation are not complete, but there might be some code you can steal. For the MeArm the ZYY ikSolver will work as is. It might work for you too, it just depends on the angles of your joints relative to each other.

Also check out https://github.com/rockbot/vektor There are other linear algebra libraries that will work, but Vektor has exactly the things you need and nothing more.

commit 010d4963 (2018-01-15): Offset was being applied twice when to() was called with a time

Move application of offset so that is only applied once.

Move update of this.value to controller update methods. A little less DRY, but having them be more proximate to the actual servoWrite makes sense.

Fixes #1429
closed 2018-01-15 by rwaldron