rwaldron/johnny-five #1431: Unable to read digital on pin 2 if reading analog on A2

dirkjacobs, 2017-12-30:
Using Arduino UNO rev 3 - Standard Firmata
If reading from analog "A2", I'm unable to read from digital pin : 2
Tested by pulling digital pin 2 to ground (configured as pull-up)

Code
====
 
let five = require("johnny-five");
let board = new five.Board({
    port: "COM3",
});

board.on("ready", function() {
    console.log('Board ready');
    button1 = new five.Button({
        pin: "2",
        isPullup: true,
    });
    
    button1.on("down", () => {
        console.log('Button pressed');
    });

    // Analog
   // Disable this code to make the digital input work
    let analog1 = new five.Sensor({
        pin: "A2", 
        freq: 100, 
        threshold: 50
    });
    
});

dirkjacobs, 2017-12-30:
Related to https://github.com/rwaldron/johnny-five/issues/1234 ? (which is posted in 2016 and still open)

rwaldron, 2018-01-04:
Don't use a string: 

```js
    button1 = new five.Button({
        pin: "2", // <-- Change this to a number, not a string. 
        isPullup: true,
    });
```

closed 2018-01-04 by rwaldron