rwaldron/johnny-five #1435: i got this error message and my app crash

mikeGHQ, 2018-01-05:
i have a very simple nwjs app that launch a script on start, this is my code:

`var five = require('johnny-five');
var board = new five.Board();

board.on('ready', function() {
  var led = new five.Led(12);
  led.blink(500);
});
`
when i start the app, some arduino built-in leds blink then in the console i get this error:

`Error [ERR_UNHANDLED_ERROR]: Unhandled error. ([object Object])
    at Board.emit (events.js:140:19)
    at Board.log (/home/mohammed/node_modules/johnny-five/lib/board.js:630:8)
    at Board.(anonymous function) [as error] (/home/mohammed/node_modules/johnny-five/lib/board.js:641:14)
    at Board.<anonymous> (/home/mohammed/node_modules/johnny-five/lib/board.js:385:14)
    at ontimeout (timers.js:466:11)
    at tryOnTimeout (timers.js:304:5)
    at Timer.listOnTimeout (timers.js:264:5)
Error: async hook stack has become corrupted (actual: 9, expected: 10)
 1: node::Environment::AsyncHooks::pop_async_id(double) [/home/mohammed/Desktop/node-webkit-projects/app/node_modules/nw/nwjs/lib/libnode.so]
 2: node::InternalCallbackScope::Close() [/home/mohammed/Desktop/node-webkit-projects/app/node_modules/nw/nwjs/lib/libnode.so]
 3: node::InternalMakeCallback(node::Environment*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*, node::async_context) [/home/mohammed/Desktop/node-webkit-projects/app/node_modules/nw/nwjs/lib/libnode.so]
 4: 0x7fcfebbdf2e1 [/home/mohammed/Desktop/node-webkit-projects/app/node_modules/nw/nwjs/lib/libnode.so]
 5: 0x7fcfebff2056 [/home/mohammed/Desktop/node-webkit-projects/app/node_modules/nw/nwjs/lib/libnode.so]
 6: uv_run [/home/mohammed/Desktop/node-webkit-projects/app/node_modules/nw/nwjs/lib/libnode.so]
 7: 0x7fcfffd23cd3 [/home/mohammed/Desktop/node-webkit-projects/app/node_modules/nw/nwjs/lib/libnw.so]`

i am using arduino mega 2560, and i uploaded a stranded firmata

mikeGHQ, 2018-01-06:
i solved the problem using a module called `nwjs-j5-fix` it is used to let you use j5 in nwjs
# the code at the first line

`nw.require("nwjs-j5-fix").fix();`

dtex, 2018-01-06:
Awesome! Thanks for following up and letting us know.

closed 2018-01-06 by dtex