rwaldron/johnny-five #1440: Connecting to a membrane 3x4 Keypad without I2C?

eman6576, 2018-01-23:
Hello,

I am working on a project that uses an Adafruit 3x4 membrane keyboard on a Raspberry Pi 3 Model B+. I read the documentation in regards to `12 Button Keypad` but it says that it requires I2C backpack. Does that only apply to Arduino boards? If so, is there another way to read input from the keypad using the `Board` class methods for interacting with pins?

dtex, 2018-01-25:
Hi @eman6576 

Sorry nobody has responded. I'm not familiar with the membrane keyboard or the Raspberry Pi, but I think I can hazard a guess (and it's just a guess).

The keypad requires tight timing when checking the pins. The commands required to toggle the pin state, check the values, and switch the pins back cannot be executed quickly enough over firmata/serialport. This limitation may not exist on the Pi, but it would stop the functionality from being added to Johnny-Five since firmata + firmata.js leads the way with functionality and drives where we go with the public API.

Maybe it's possible to do with rasp-io. I'm really not sure. Maybe it's possible to go even lower and write something custom. Have you seen any examples (besides the i2c backpack) where the membrane keyboard is used in node? What are they using?

eman6576, 2018-01-26:
@dtex The way that I am doing it currently is running a separate program in python that polls the keypad and prints to stdout the key entered. Then I am using [python-shell](https://www.npmjs.com/package/python-shell) to read in the output and handle it in Node. But I haven't seen anyone that has tried to do it in Node. But rasp-io might be the best place to add the functionality.

dtex, 2018-01-26:
Neat. Is the python program online somewhere? I'd like to see how it works.

eman6576, 2018-01-26:
The project is closed source but I could post a sample project if you want to take a look.

rwaldron, 2018-02-20:
Membrane keypads like the one described here require synchronous digital pin read and write operations that  interleave in nested loops (for each row, for each column) to determine "currently pressed button": https://github.com/rwaldron/johnny-five/blob/master/firmwares/keypad_3x4_i2c_nano_backpack.ino#L91-L107 This simply won't work with Node.js IO

closed 2018-02-20 by rwaldron

eman6576, 2018-02-20:
@rwaldron Yes I have tried that approach but the performance is terrible. In Python, you can use event handlers for detecting when a button was tapped which is better than using loops. It would be cool if that could be done here.

mpacary, 2020-03-04:
Hello,

I could get [this 4x4 membrane keypad](https://www.cytron.io/p-4x4-matrix-membrane-keypad), directly wired on 8x RasPi GPIO pins, to work with Johnny-Five.

Setup :

* Hard : Raspberry Pi 4B, 16x2 LCD display behind PCF8574T, 4x4 membrane keypad

* Soft : Node v8 (I know it's old but needed for other lib compatibility ; should work with most recent versions however), Johnny-Five 1.4, Raspi-IO 11.0

### **[Source](https://gist.github.com/mpacary/e0ee252404927c1dd4346b2e8e01ae64)**

Open to comments / improvements, I'm a beginner in the IoT field :baby: 

Hopefully this will give help for an official Johnny-Five support of this kind of material & wiring.