rwaldron/johnny-five #1441: On executing first.js from php in owner user I get board closed.

arzel33, 2018-02-01:
I want to run a webapp which controls the arduino in raspberry (LAMP), so I started developing this app but on the first step I get an error, it wasn't initializing the program so I changed the user php was executing commands, then the program is being executed but johnny-five is closing board immediately so it's impossible to run first.js properly. Does anyone know what can I do to fix it?

dtex, 2018-02-01:
@arzel33 I'm not a Pi expert, but I know whoever helps you will need more info. Can you post or link to the code you are running and show us what is output to the conosole when you run the code?

arzel33, 2018-02-01:
That's php running after ajax connection: `if (isset($_POST["position_off"])) {
	$command = shell_exec("node /home/pi/Projects/raspDuino/tests/first.js");
}
	echo json_encode($command);`
And that's node.js: ` var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
	console.log("Ready!");
	var led = new five.Led(8);

	this.repl.inject({
		led: led
	});
	led.blink(500);
});
`
If I run this file it's working at nomal way but in the way I want it's not working.

dtex, 2018-02-01:
Do you get anything back in ```$command```? That could be helpful.

arzel33, 2018-02-02:
@dtex   Yes, I get: 1-Board connected; 2-Initializing; 3-Ready (the console.log); 4-Board Closing;
That's literally what I get.

dtex, 2018-02-02:
I'm just speculating, but I wonder if your attempt to use the REPL is at the root of the problem. Can you try this code:

````
var five = require("johnny-five");
var board = new five.Board({ repl: false });
board.on("ready", function() {
	led.blink(500);
});
````

arzel33, 2018-02-02:
Now it works, thanks a lot!! Could you tell me, please why I was getting this error? And why I can only use repl giving commands on normal terminal? Thanks a lot!!

dtex, 2018-02-02:
Again, I'm just guessing, but I figure that when you use shell_exec you don't get the same functionality from node's ```Readline.Interface``` which the repl depends on.

It would be nice if it gave a meaningful error on the console but this is such an unusual case.

I'm curious, why would you need a repl if you are running the program from a shell in PHP? It seems a little convoluted and I wonder if maybe there is an easier way to achieve your desired functionality.

arzel33, 2018-03-27:
Hi! @dtex sorry for the waiting. What I wanted to do with repl was sync webapp and nodeJs and avoid doing it with a websocket. I had an other problem, when you open the board and the script, the board keeps alive but the script not so timeouts an other stuff will be useless. If someone have this problem... I fixed it using [forever package](https://www.npmjs.com/package/forever).

dtex, 2018-03-27:
>  If someone have this problem... I fixed it using forever package.
Cool, thanks for noting that. 

> avoid doing it with a websocket
I've always just used Websockets (socket.io) for having a web app talk to my J5 app.

It sounds like I can close this. If there are any other issues feel free to re-open, or open a new one.

Cheers

closed 2018-03-27 by dtex