rwaldron/johnny-five #1446: Could you provide MPU-9250 IMU support ?

fabienberteau, 2018-02-11:
Hello,

I have inherited a [Grove IMU 9DOF v](http://wiki.seeed.cc/Grove-IMU_9DOF_v2.0/) and i would wanted to use it with your awesome Johnny-five framework. Could you help me ?

Thanks

Fabien

rwaldron, 2018-04-05:
Sorry about the delay, I've been in the process of moving office and as a new dad I'm generally behind on everything 😉 

I'd also like to see this supported, so I will go ahead and order the component

DavideBecker, 2018-06-06:
The Digital Motion Processor of the MPU-9250 is pretty awesome, as it supports sensor fusion with quaternion calculation etc, but it was a pain to get it running. The [sparkfun library](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library) worked really well for me. Might be useful for development

dtex, 2019-06-01:
I've added the request for MPU-9250 to the [requested features](https://github.com/rwaldron/johnny-five/wiki/Requested-Features) wiki page so that it does not get forgotten.

closed 2019-06-01 by dtex