rwaldron/johnny-five #1448: how to re-open the serial port by using johnny-five by re-deploying the flow in node-red?

sidddharth7890, 2018-02-22:
hello,
right now i am using johnny-five with node-red and the problem is when i connect my arduino uno board first time it works fine, but when i re-deploy the flow in Node-red then it shows an error "Port is not open!"

**running the flow first time:**
 
Welcome to Node-RED
===================

22 Feb 09:02:31 - [info] Node-RED version: v0.18.2
22 Feb 09:02:31 - [info] Node.js  version: v4.2.1
22 Feb 09:02:31 - [info] Windows_NT 10.0.16299 x64 LE
22 Feb 09:02:32 - [info] Loading palette nodes
22 Feb 09:02:33 - [info] Dashboard version 2.7.0 started at /ui
22 Feb 09:02:33 - [warn] ------------------------------------------------------
22 Feb 09:02:33 - [warn] [node-red/rpi-gpio] Info : Ignoring Raspberry Pi specific node
22 Feb 09:02:33 - [warn] [node-red/tail] Not currently supported on Windows.
22 Feb 09:02:33 - [warn] ------------------------------------------------------
22 Feb 09:02:33 - [info] Settings file  : \Users\om\.node-red\settings.js
22 Feb 09:02:33 - [info] User directory : \Users\om\.node-red
22 Feb 09:02:33 - [warn] Projects disabled : set editorTheme.projects.enabled=true to enable
22 Feb 09:02:33 - [info] Flows file     : \Users\om\.node-red\flows_DESKTOP-L5ID322.json
22 Feb 09:02:33 - [info] Server now running at http://127.0.0.1:1880/
22 Feb 09:02:33 - [info] Starting flows
1519270353476 Connected COM6
22 Feb 09:02:33 - [info] Started flows
1519270354301 Repl Initialized

**After Re-deploying the flow inside Node-red:**

Welcome to Node-RED
===================

22 Feb 09:02:31 - [info] Node-RED version: v0.18.2
22 Feb 09:02:31 - [info] Node.js  version: v4.2.1
22 Feb 09:02:31 - [info] Windows_NT 10.0.16299 x64 LE
22 Feb 09:02:32 - [info] Loading palette nodes
22 Feb 09:02:33 - [info] Dashboard version 2.7.0 started at /ui
22 Feb 09:02:33 - [warn] ------------------------------------------------------
22 Feb 09:02:33 - [warn] [node-red/rpi-gpio] Info : Ignoring Raspberry Pi specific node
22 Feb 09:02:33 - [warn] [node-red/tail] Not currently supported on Windows.
22 Feb 09:02:33 - [warn] ------------------------------------------------------
22 Feb 09:02:33 - [info] Settings file  : \Users\om\.node-red\settings.js
22 Feb 09:02:33 - [info] User directory : \Users\om\.node-red
22 Feb 09:02:33 - [warn] Projects disabled : set editorTheme.projects.enabled=true to enable
22 Feb 09:02:33 - [info] Flows file     : \Users\om\.node-red\flows_DESKTOP-L5ID322.json
22 Feb 09:02:33 - [info] Server now running at http://127.0.0.1:1880/
22 Feb 09:02:33 - [info] Starting flows
1519270353476 Connected COM6
22 Feb 09:02:33 - [info] Started flows
1519270354301 Repl Initialized
>> 22 Feb 09:03:27 - [info] Stopping flows
board closed
22 Feb 09:03:27 - [info] Stopped flows
22 Feb 09:03:27 - [info] Starting flows
1519270407709 Connected COM6
22 Feb 09:03:27 - [info] Started flows
Opening \\.\COM6: Access denied
Port is not open
Port is not open

rwaldron, 2018-02-26:
@monteslu any advice for this?

monteslu, 2018-02-28:
This is a strange one.  I'm explicitly telling serialport not to lock:  https://github.com/monteslu/node-red-contrib-gpio/blob/master/lib/nodebotNode.js#L131
Though I haven't really done much windows testing.  I'll try to reproduce later this week.

sidddharth7890, 2018-03-04:
@rwaldron  any updates?? it is really problematic

rwaldron, 2018-03-05:
@sidddharth7890 no, I don't

monteslu, 2018-03-06:
@sidddharth7890 I was able to duplicate it today.  Found the problem, but not sure just how to fix it yet.  

The problem is that node-red-contrib-gpio isn't waiting for the serialport.close() method to finish before restarting the flow.  Serialport.close() can take a callback, but will require a bit of refactoring of  this code https://github.com/monteslu/node-red-contrib-gpio/blob/master/lib/nodebotNode.js#L38-L100  to utilize the callback.  Definitely open to any PRs on that :)

@rwaldron This is a node-red-contrib-gpio problem not a johnny-five issue, so we should close this one.

sidddharth7890, 2018-03-07:
thanks for your response

dtex, 2018-03-07:
Closing this. Please see new issue: https://github.com/monteslu/node-red-contrib-gpio/issues/30

Thanks @monteslu, hope to see you at a conference soon. I think I owe you some pizza and beer :-)

closed 2018-03-07 by dtex

rwaldron, 2018-03-07:
@monteslu thank you sir!