rwaldron/johnny-five #1450: node-red-contrib-gpio problems

Waterbrain, 2018-02-25:
Hello:
I just installed node red from scratch in a raspberry with stretch and apart from the problem that I will continue to explain, when I install the package, it deactivates, unlinks or whatever the node-red-dashboard package, although the latter happens with other packages .
The fact is that node-network stops working and lists these messages:

Welcome to Node-RED 
=================== 
25 Feb 10:29:51 - [info] Node-RED version: v0.18.3 
25 Feb 10:29:51 - [info] Node.js  version: v8.9.4 
25 Feb 10:29:51 - [info] Linux 4.9.59-v7+ arm LE 
25 Feb 10:29:53 - [info] Loading palette nodes 
25 Feb 10:29:58 - [info] RedBot version: 0.9.23 (node-red-contrib-chatbot) 
25 Feb 10:30:13 - [info] Dashboard version 2.8.2-beta started at /ui 
25 Feb 10:30:16 - [info] Settings file  : /home/pi/.node-red/settings.js 
25 Feb 10:30:16 - [info] User directory : /home/pi/.node-red 
25 Feb 10:30:16 - [warn] Projects disabled : set editorTheme.projects.enabled=true to enable 
25 Feb 10:30:16 - [info] Flows file     : /home/pi/.node-red/flows_invernadero.json 
25 Feb 10:30:16 - [info] Server now running at http://127.0.0.1:1880/ 
25 Feb 10:30:17 - [info] Starting flows 
1519551017130 Available Firmata 
1519551017312 Available RaspberryPi-IO 
25 Feb 10:30:17 - [info] Started flows 
2018-02-25 10:30:17 initCheckPermitted: 
+---------------------------------------------------------+ 
|Sorry, you don't have permission to run this program.    | 
|Try running as root, e.g. precede the command with sudo. | 
+---------------------------------------------------------+ 
25 Feb 10:30:17 - [red] Uncaught Exception: 
25 Feb 10:30:17 - Error: pigpio error -1 in gpioInitialise 
    at initializePigpio (/home/pi/.node-red/node_modules/pigpio/pigpio.js:11:12) 
    at new Gpio (/home/pi/.node-red/node_modules/pigpio/pigpio.js:25:3) 
    at new DigitalOutput (/home/pi/.node-red/node_modules/raspi-gpio/dist/index.js:79:25) 
    at RaspiIOCore.value (/home/pi/.node-red/node_modules/raspi-io-core/dist/index.js:553:38) 
    at RaspiIOCore.pinMode (/home/pi/.node-red/node_modules/raspi-io-core/dist/index.js:505:21) 
nodered.service: Main process exited, code=exited, status=1/FAILURE 
nodered.service: Unit entered failed state. 
nodered.service: Failed with result 'exit-code'. 
nodered.service: Service hold-off time over, scheduling restart. 
Stopped Node-RED graphical event wiring tool.. 
Started Node-RED graphical event wiring tool.. 
1519551022858 Available /dev/ttyUSB0 
1519551022925 Connected /dev/ttyUSB0 
25 Feb 10:30:23 - [info] 
Welcome to Node-RED 
=================== 

Can somebody help me?
Thank you very much.

rwaldron, 2018-02-26:
The answer you're looking for is in the output shown above: 

```
+---------------------------------------------------------+ 
|Sorry, you don't have permission to run this program.    | 
|Try running as root, e.g. precede the command with sudo. | 
+---------------------------------------------------------+ 
25 Feb 10:30:17 - [red] Uncaught Exception: 
25 Feb 10:30:17 - Error: pigpio error -1 in gpioInitialise 
    at initializePigpio (/home/pi/.node-red/node_modules/pigpio/pigpio.js:11:12) 
    at new Gpio (/home/pi/.node-red/node_modules/pigpio/pigpio.js:25:3) 
    at new DigitalOutput (/home/pi/.node-red/node_modules/raspi-gpio/dist/index.js:79:25) 
    at RaspiIOCore.value (/home/pi/.node-red/node_modules/raspi-io-core/dist/index.js:553:38) 
    at RaspiIOCore.pinMode (/home/pi/.node-red/node_modules/raspi-io-core/dist/index.js:505:21) 
nodered.service: Main process exited, code=exited, status=1/FAILURE 
nodered.service: Unit entered failed state. 
nodered.service: Failed with result 'exit-code'. 
nodered.service: Service hold-off time over, scheduling restart. 
```

There is a problem with the permissions of the user invoking the script. Have you tried running it as `sudo`? @nebrius may also have some advice.

nebrius, 2018-02-26:
Yep, Raspi IO must be run as root, as documented in the [usage section of the Raspi IO README](https://github.com/nebrius/raspi-io#usage). This is a requirement put on us by Raspbian itself, so there's nothing we can do about it unfortunately.

Try running:

```
sudo node-red-start
```

Waterbrain, 2018-02-27:
Thank you very much to all.

The way to start that job is:

```
sudo service nodered start
```
From the .node-red directory.

Since if I do it as user pi:

```
pi@invernadero:~/.node-red $ service nodered start
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Se requiere autenticación para iniciar 'nodered.service'.
Multiple identities can be used for authentication:
 1.  ,,, (pi)
 2.  root
Choose identity to authenticate as (1-2): 1
Password: 
polkit-agent-helper-1: pam_authenticate failed: Authentication failure
==== AUTHENTICATION FAILED ===
Failed to start nodered.service: Access denied
See system logs and 'systemctl status nodered.service' for details.
```
If I do this:

```
pi@invernadero:~/.node-red $ node-red-start 

Start Node-RED
 
Once Node-RED has started, point a browser at http://192.168.1.102:1880
On Pi Node-RED works better with the Firefox or Chrome browser
 
Use   node-red-stop                          to stop Node-RED
Use   node-red-start                         to start Node-RED again
Use   node-red-log                           to view the recent log output
Use   sudo systemctl enable nodered.service  to autostart Node-RED at every boot
Use   sudo systemctl disable nodered.service to disable autostart on boot
 
To find more nodes and example flows - go to http://flows.nodered.org
 
Starting as a systemd service.
Started Node-RED graphical event wiring tool..
1519716224883 Available /dev/ttyUSB0
1519716224949 Connected /dev/ttyUSB0
27 Feb 08:23:45 - [info]
Welcome to Node-RED
===================
27 Feb 08:23:45 - [info] Node-RED version: v0.18.3
27 Feb 08:23:45 - [info] Node.js  version: v8.9.4
27 Feb 08:23:45 - [info] Linux 4.9.59-v7+ arm LE
27 Feb 08:23:47 - [info] Loading palette nodes
27 Feb 08:23:52 - [info] RedBot version: 0.9.23 (node-red-contrib-chatbot)
27 Feb 08:24:06 - [info] Dashboard version 2.8.2-beta started at /ui
27 Feb 08:24:09 - [info] Settings file  : /home/pi/.node-red/settings.js
27 Feb 08:24:09 - [info] User directory : /home/pi/.node-red
27 Feb 08:24:10 - [warn] Projects disabled : set editorTheme.projects.enabled=true to enable
27 Feb 08:24:10 - [info] Flows file     : /home/pi/.node-red/flows_invernadero.json
27 Feb 08:24:10 - [info] Server now running at http://127.0.0.1:1880/
27 Feb 08:24:10 - [info] Starting flows
1519716250627 Available Firmata
1519716250816 Available RaspberryPi-IO
27 Feb 08:24:10 - [info] Started flows
2018-02-27 08:24:10 initCheckPermitted:
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+
27 Feb 08:24:10 - [red] Uncaught Exception:
27 Feb 08:24:10 - Error: pigpio error -1 in gpioInitialise
nodered.service: Main process exited, code=exited, status=1/FAILURE
nodered.service: Unit entered failed state.
nodered.service: Failed with result 'exit-code'.
nodered.service: Service hold-off time over, scheduling restart.
Stopped Node-RED graphical event wiring tool..
Started Node-RED graphical event wiring tool..
````
And:

````
pi@invernadero:~/.node-red $ sudo node-red-start 

Start Node-RED
 
Once Node-RED has started, point a browser at http://192.168.1.102:1880
On Pi Node-RED works better with the Firefox or Chrome browser
 
Use   sudo systemctl enable nodered.service  to autostart Node-RED at every boot
Use   sudo systemctl disable nodered.service to disable autostart on boot
 
To find more nodes and example flows - go to http://flows.nodered.org
27 Feb 08:25:12 - [info] 

Welcome to Node-RED
===================

27 Feb 08:25:12 - [info] Node-RED version: v0.18.3
27 Feb 08:25:12 - [info] Node.js  version: v8.9.4
27 Feb 08:25:12 - [info] Linux 4.9.59-v7+ arm LE
27 Feb 08:25:14 - [info] Loading palette nodes
27 Feb 08:25:20 - [info] Settings file  : /root/.node-red/settings.js
27 Feb 08:25:20 - [info] User directory : /root/.node-red
27 Feb 08:25:20 - [warn] Projects disabled : set editorTheme.projects.enabled=true to enable
27 Feb 08:25:20 - [info] Flows file     : /root/.node-red/flows_invernadero.json
27 Feb 08:25:20 - [info] Server now running at http://127.0.0.1:1880/
27 Feb 08:25:21 - [info] Waiting for missing types to be registered:
27 Feb 08:25:21 - [info]  - ui_base
27 Feb 08:25:21 - [info]  - ui_tab
27 Feb 08:25:21 - [info]  - ui_group
27 Feb 08:25:21 - [info]  - nodebot
27 Feb 08:25:21 - [info]  - gpio out
27 Feb 08:25:21 - [info]  - ui_gauge
27 Feb 08:25:21 - [info]  - ui_chart
27 Feb 08:25:21 - [info]  - ui_template
27 Feb 08:25:21 - [info]  - ui_dropdown
27 Feb 08:25:21 - [info]  - ui_switch
27 Feb 08:25:21 - [info]  - ui_button
27 Feb 08:25:21 - [info]  - ui_slider
27 Feb 08:25:21 - [info]  - rpi-srf
27 Feb 08:25:21 - [info]  - ui_text
27 Feb 08:25:21 - [info]  - ui_audio
```

Node-red starts but loses the dashboard ...

Waterbrain, 2018-02-27:
nebrius npm i raspi-io also have to install it as sudo?

closed 2019-03-01 by stale[bot]