rwaldron/johnny-five #1452: Problem with detecting device using Windows subsystem for Linux

JoarGjersund, 2018-03-10:
I have connected Arduino Uno to my pc and uploaded the StandardFirmdataPlus, but whenever i try to run a script it is stuck at Board Looking for connected device. Anyone know what the problem is?

coolhandfluke, 2018-03-10:
Check in windows device manager, under ports what is showing, if nothing is showing you need to install the correct driver, once it appears there then you can upload standard firmata plus and run the script, in the script you can also specify the exacy port like new Board({port: 'COM3' }) although this is not mandatory, update here what happened, once you did this.

JoarGjersund, 2018-03-10:
The board is connected (and visible as COM1 in task manager) as I said, the firmdata is allready uploaded, but I'm not able to recognize it in ubuntu. I tired specifying COM1 like you said, I just get: 
Connected COM1
1520697568727 Error Error: No such file or directory, cannot open COM1

coolhandfluke, 2018-03-10:
I never used windows subsystem for Linux, sorry I can't help with your issue.  I have Ubuntu VM in my windows pc, we're you able to run any Johnny five scripts directly on windows? I tried this but getting a different error which I mentioned in issue #1451

JoarGjersund, 2018-03-11:
Looks like this is a WSL issue. https://github.com/Microsoft/WSL/issues/412

closed 2018-03-11 by JoarGjersund