rwaldron/johnny-five #1457: HC-SR04 I2C Backpack with Tessel 2 - Socket: Error: write EPIPE

JeanMichelBoeuf, 2018-03-31:
Hi everyone !

I'm trying to use HC-SR04 ultrasonic sensor with Tessel 2 through an I2C backpack, unsuccessfully for the moment. I can't have any reading from the sensor :

the 1st time I launch the test program (available below ), I have the following error:

```
1522525570883 Available Tessel 2 (MrBot)
1522525571065 Connected Tessel 2 (MrBot)
1522525571147 Repl Initialized

Socket: Error occurred: Error: write EPIPE
/usr/lib/node/tessel-export.js:337
        throw new Error('Socket: The Port socket has closed.');
        ^

Error: Socket: The Port socket has closed.
    at Socket.Port.sock.on (/usr/lib/node/tessel-export.js:337:15)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:497:12)
````
Just after that I'm lauching it again, no error this time but no value either :
````
Proximity:
  cm  :  0
  in  :  0
-----------------
````
**Since the Port A led is turned off, it's not surprising that it doesn't work. There's no current in any of the A ports.**

The material I'm using :

* Tessel 2
* arduino nano (installed as I2C backpack with "*interchange install hc-sr04 -a uno*")
* HC-SR04
* SparkFun Logic Level Converter - Bi-Directional

[image]

The test code :
````
const five = require('johnny-five');
const Tessel = require('tessel-io');

const board = new five.Board({
  io: new Tessel(),
});

board.on('ready', () => {

  const proximity = new five.Proximity({
    controller: "HCSR04I2CBACKPACK",
  });

  proximity.on('data', function () {
    console.log('Proximity: ');
    console.log('  cm  : ', this.cm);
    console.log('  in  : ', this.in);
    console.log('-----------------');
  });

  proximity.on('change', () => {
    console.log('The obstruction has moved.');
  });
});
````

What I have tested so far :

* Arduino nano is working
* HC-SR04 is working; tested with arduino nano only and pingFirmata (installed with "interchange install hc-sr04 -a uno --firmata")
* I2C backpack install confirmed with no error during installation then "*interchange read -p /dev/ttyUSB0*"
* Tessel 2 ports A have been tested with another project.
**weird behavior : I have tried to use I2C on port B (B0 and B1). Surprisingly there's no error and Port A led is light on and testing port a1 with a multimter read 3.3V.**

Maybe it's a Tessel 2 issue and not Johnny-five but I don't have the level to judge that.
Any help would be greatly appreciate, I'm quite lost now.

rwaldron, 2018-04-04:
> Error: Socket: The Port socket has closed.

This is usually caused by some failure that crashes the actual port (Port A or B). Can you try this: 

```js
const five = require('johnny-five');
const Tessel = require('tessel-io');

const board = new five.Board({
  io: new Tessel(),
});

board.on('ready', () => {
  console.log('ready!');
});
```

And report back the outcome? Thanks!

JeanMichelBoeuf, 2018-04-12:
I've finally been able  to test it, so here is the result : 
```
INFO Looking for your Tessel...
INFO Connected to MrBot.
INFO Building project.
INFO Writing project to RAM on MrBot (1080.32 kB)...
INFO Deployed.
INFO Running componentTest/proximity.js...
1523568442146 Available Tessel 2 (MrBot)
1523568442322 Connected Tessel 2 (MrBot)
1523568442400 Repl Initialized
>> ready!
```

closed 2019-04-14 by stale[bot]