rwaldron/johnny-five #1458: GPIO error creating a new pin instance on a Beaglebone Black in Debian 9.2

totem415, 2018-04-04:
I'm on a Beaglebone Black, Debian 9.2, Node JS v6.11.4, running the latest beaglebone-io / johnny five combo.

I'm getting this error when I create a new pin instance to control a GPIO to use as a digital in.

Here's my standalone app code:

```
var five = require('johnny-five');
var BeagleBone = require('beaglebone-io');

var board = new five.Board({
    io: new BeagleBone()
});

board.on("ready", function() {
    var mechRprtReadyPin = new five.Pin({
        pin: "P9_23",
        type: 'digital',
        mode: 0
    });
    mechRprtReadyPin.on('high', onMechRprtPinHigh);
    mechRprtReadyPin.on('low', onMechRprtPinLow);
    console.log("Pin Control initialization complete.");
});

function onMechRprtPinHigh(data) {
    console.log('PIN: HIGH (mechaduino report event)  data: ' + JSON.stringify(data));
}

function onMechRprtPinLow(data) {
    console.log('Pin: LOW  data: ' + JSON.stringify(data));
}
```

And here's the output / call stack of the error:

```
root@1UpTester:/var/lib/cloud9/md/trunk# node gpioTest
1522802125440 Available BeagleBone-IO
1522802125585 Connected BeagleBone-IO
1522802125782 Repl Initialized
>> fs.js:642
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                 ^

Error: ENOENT: no such file or directory, open '/sys/devices/platform/ocp//ocp:P9_23_pinmux/state'
    at Error (native)
    at Object.fs.openSync (fs.js:642:18)
    at Object.fs.writeFileSync (fs.js:1348:33)
    at Object.module.exports.state (/var/lib/cloud9/md/trunk/node_modules/beaglebone-io/lib/pinmux.js:9:6)
    at BeagleBone._pinModeInput (/var/lib/cloud9/md/trunk/node_modules/beaglebone-io/lib/beaglebone.js:34:10)
    at BeagleBone.LinuxIO.pinMode (/var/lib/cloud9/md/trunk/node_modules/linux-io/lib/linux-io.js:99:12)
    at Pin.set (/var/lib/cloud9/md/trunk/node_modules/johnny-five/lib/pin.js:82:17)
    at new Pin (/var/lib/cloud9/md/trunk/node_modules/johnny-five/lib/pin.js:90:13)
    at Board.<anonymous> (/var/lib/cloud9/md/trunk/gpioTest.js:9:28)
    at emitNone (events.js:91:20)
```

The error is generated on this line:
```
var mechRprtReadyPin = new five.Pin({
        pin: "P9_23",
        type: 'digital',
        mode: 0
    });
```

Any help would be very much appreciated.

dtex, 2018-04-04:
Not to brush you off, but you might get more timely help over at https://github.com/julianduque/beaglebone-io

They are the beaglebone-io experts

fivdi, 2018-04-04:
I'll take a look at this later here when I have some time to spare.

fivdi, 2018-04-04:
This issue has moved over to https://github.com/julianduque/beaglebone-io/issues/66 so it can be closed here in J5 repository.

closed 2018-04-04 by totem415