rwaldron/johnny-five #1467: LCD Cursor

Spy474, 2018-05-05:
I don't know if I'm asking at the Right place, if not, let me know!

So, I'm trying to learn Johnny Five with an Arduino Uno. My goal is to learn how to control a LCD screen (20x04). So far, it's almost all good!

My only concern for now, I'm trying to display a Menu, and I would like to be able to remove the Cursor. I try looking online and I cannot find anything!

Thank you for your help!

fivdi, 2018-05-05:
The [noCursor](https://github.com/rwaldron/johnny-five/blob/v0.14.3/lib/lcd.js#L1008-L1015) method which doesn't appear to be documented may help.

Spy474, 2018-05-05:
I already tried the noCursor function. It remove all the "layout" that I would have set, and the cursor is still there. Not sure how to use it!

fivdi, 2018-05-05:
Can you post a short program that can be used to reproduce the problem please?

fivdi, 2018-05-06:
@Spy474 I just tried the below program on a 16x02 parallel LCD with an Arduino UNO and it worked as expected. I does the following:

- Displays "Hello, World!" (the cursor is off by default)
- Turns the cursor on after two seconds
- Starts blinking the cursor after four seconds
- Stops blinking the cursor after six seconds
- Turns the cursor off after eight seconds

 I don't have a 5V 20x04 LCD to test with.

```js
const five = require("johnny-five");
const board = new five.Board();

board.on("ready", () => {

  var lcd = new five.LCD({ pins: [7, 8, 9, 10, 11, 12] });

  lcd.print("Hello, World!");

  setTimeout(() => {
    lcd.cursor();
  }, 2000);

  setTimeout(() => {
    lcd.blink();
  }, 4000);

  setTimeout(() => {
    lcd.noBlink();
  }, 6000);

  setTimeout(() => {
    lcd.noCursor();
  }, 8000);
});
```

Spy474, 2018-05-06:
@fivdi I just tried your exact program and the result are totally different:

So, just to make sure everything is right:
Node version: v8.11.1
+ johnny@2.0.0
+ five@0.8.0

Displays "Hello, World!" (the cursor is on by default for me)
Turns the cursor on after two seconds (Doesn't change anything)
Starts blinking the cursor after four seconds (No Blink)
Stops blinking the cursor after six seconds (No Blink)
Turns the cursor off after eight seconds (Stay on)

Should I change anything from the original file in Johnny Five? Can I tell the software that I'm working with 20x04?

I think the problem come from the controller. I'll try to find in the script, but I'm not to familiar with coding, trying to figure it out!

Thank you so much for your help!

`const five = require("johnny-five");
const board = new five.Board();

board.on("ready", () => {

  var lcd = new five.LCD({ controller: "PCF8574A" });
  lcd.print("Hello, World!");

  setTimeout(() => {
    lcd.cursor();
  }, 2000);

  setTimeout(() => {
    lcd.blink();
  }, 4000);

  setTimeout(() => {
    lcd.noBlink();
  }, 6000);

  setTimeout(() => {
    lcd.noCursor();
  }, 8000);
});`

fivdi, 2018-05-06:
I used Node v8.10.0 and johnny-five@0.14.3. 

> - johnny@2.0.0
> - five@0.8.0

`johnny` and `five` are not the correct packages to use. Were they really used?

For a 20x04 display the display size should be specified when creating a new LCD instance:
```js
var lcd = new five.LCD({ controller: "PCF8574A", cols: 20, rows: 4 });
```
Try with this to see if it helps.

You're using an I2C LCD. I'm afraid I don't have one of those to perform any tests.

LarsVanderheydt, 2019-03-23:
@Spy474 Had the same issue using a I2C LCD (2x16).
Ended up using this.

`lcd.cursor(1, 20);`

Just tell it to go out of screen.
Hope this helps.

closed 2020-03-24 by stale[bot]

rudemex, 2020-04-11:
For a 20x4 display with I2C (PCF8574T) I operate with the command
```
lcd.cursor(0, 49)
```